RPi-Distro/pi-gen #499: FATAL: Module nbd not found in directory /lib/modules/5.11.6-arch1-1 (using docker)

pirateradiohack, 2021-03-19:
While trying to build the image with `./build-docker.sh`, it used to work fine. But now I get this error after configuring packages:

```
...
Setting up wget (1.20.1-1.1) ...
Setting up libarchive13:i386 (3.3.3-4+deb10u1) ...
Setting up bc (1.07.1-2+b1) ...
Setting up libldap-2.4-2:i386 (2.4.47+dfsg-3+deb10u6) ...
Setting up kmod (26-1) ...
Setting up ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
137 added, 0 removed; done.
Setting up bsdmainutils (11.1.2+b1) ...
update-alternatives: using /usr/bin/bsd-write to provide /usr/bin/write (write) in auto mode
update-alternatives: using /usr/bin/bsd-from to provide /usr/bin/from (from) in auto mode
Setting up libgssapi-krb5-2:i386 (1.17-3+deb10u1) ...
Setting up udev (241-7~deb10u6) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up libcroco3:i386 (0.6.12-3) ...
Setting up libgdbm-compat4:i386 (1.18.1-4) ...
Setting up libperl5.28:i386 (5.28.1-6+deb10u1) ...
Setting up libcurl4:i386 (7.64.0-4+deb10u1) ...
Setting up curl (7.64.0-4+deb10u1) ...
Setting up qemu-utils (1:3.1+dfsg-8+deb10u8) ...
Setting up rsync (3.1.3-6) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of restart.
Setting up libarchive-tools (3.3.3-4+deb10u1) ...
Setting up debootstrap (1.0.114) ...
Setting up gettext (0.19.8.1-9) ...
Setting up libcurl3-gnutls:i386 (7.64.0-4+deb10u1) ...
Setting up bsdtar (3.3.3-4+deb10u1) ...
Setting up perl (5.28.1-6+deb10u1) ...
Setting up quilt (0.65-3) ...
Setting up liberror-perl (0.17027-2) ...
Setting up git (1:2.20.1-2+deb10u3) ...
Setting up libdevmapper1.02.1:i386 (2:1.02.155-3) ...
Setting up libparted2:i386 (3.2-25) ...
Setting up dmsetup (2:1.02.155-3) ...
Setting up kpartx (0.7.9-3+deb10u1) ...
Setting up parted (3.2-25) ...
Processing triggers for libc-bin (2.28-10) ...
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Removing intermediate container 0ed46896e89b
 ---> e59df3dac02b
Step 5/6 : COPY . /pi-gen/
 ---> b1451fc08b78
Step 6/6 : VOLUME [ "/pi-gen/work", "/pi-gen/deploy"]
 ---> [Warning] The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in bda034f1ad10
Removing intermediate container bda034f1ad10
 ---> 1e47a099d7fe
Successfully built 1e47a099d7fe
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
[10:29:50] Begin /pi-gen
[10:29:50] Begin /pi-gen/stage0
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.11.6-arch1-1/modules.dep.bin'
modprobe: FATAL: Module nbd not found in directory /lib/modules/5.11.6-arch1-1
[10:29:50] Unloading image

real    0m0.997s
user    0m0.037s
sys     0m0.037s

```

Did I miss something?

OrbitalTeapot, 2021-04-09:
Hi, is the USE_QCOW2 environment variable set to 0 in your config file?
More details can be found in the updated README.md

pirateradiohack, 2021-04-10:
Thanks @orbitalteapot for your help. Indeed, setting `USE_QCOW2="0"` allowed the build to succeed. 

Now I still have to understand what went wrong and why I can't build with QCOW2.

closed 2021-04-10 by pirateradiohack

pirateradiohack, 2021-11-27:
Forgot to mention the solution: I just had to load the corresponding module.
In my case on Arch Linux: `modprobe nbd`.