RPi-Distro/pi-gen #501: Add swupdate

strowk, 2021-03-26:
I think if one of the stages included swupdate, it would make updating device later easier - https://sbabic.github.io/swupdate/swupdate.html#running-swupdate

XECDesign, 2021-03-29:
I'm not getting a clear description of what it is or what it does from that page. It doesn't look like something that's needed in a general purpose OS, or maybe I'm missing something?

What's difficult about updating the device now and how does this software solve it? Keep in mind the long term plan is to add cloud-init support, which may have the functionality you want to get through swupdate.

strowk, 2021-03-29:
> SWUpdate provides a reliable way to update the software on an embedded system 

For example default implementation of azure device update agent uses it - https://github.com/Azure/iot-hub-device-update/blob/7a6e4c12d54df7a92984e935c3050f35e359a342/scripts/build.sh#L28 

If included in raspberry pi, it could probably be updated from Azure cloud.

XECDesign, 2021-04-05:
Sorry, there's not enough to go on here. I don't see how it compares to cloud-init, why updating from azure matters, or how it would be useful for the majority of users, so closing.

closed 2021-04-05 by XECDesign