RPi-Distro/pi-gen #504: Bullseye release is not available.

guillaje, 2021-04-12:
In the documentation, for the RELEASE parameter, it is said : "Valid values are jessie, stretch buster, bullseye, and testing.".
But the build of the image fails : "E: The repository 'http://archive.raspberrypi.org/debian bullseye Release' does not have a Release file."
In fact, the only releases available on http://archive.raspberrypi.org/debian/dists/ are : buster, jessie, stretch and wheezy.

Maybe the documentation should be fixed, or the script completed if a bullseye release actually exists somewhere else...

XECDesign, 2021-04-12:
Interesting timing. Bullseye has been added to the repo a few minutes ago, but don't expect to get anything resembling a usable image at this point.

The documentation is wrong if it's not along the lines of "Defaults to the currently supported release. Change this only if you know what you're doing".

guillaje, 2021-04-12:
Excellent timing ! :-D
Thank you for your quick answer.
I'll try to build some basic images (I can say I probably know what I'm doing)
I suppose you don't need feedback for the moment ? Or maybe on a mailing list or an IRC chan ?

XECDesign, 2021-04-12:
Probably a bit early for feedback before we populate the repo with bullseye packages.

free-x, 2021-05-18:
fell into a trap. raspi-config is not available
```
[19:19:09] Begin /home/user/tmp/pi-gen/stage1/03-install-packages/00-packages
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package raspi-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'raspi-config' has no installation candidate
```

pangzineng, 2021-06-05:
I just tried to build with `bullseye` and failed, I found a few packages not available:

- stage2
```
python-rpi.gpio
vl805fw
```
- stage3
```
raspberrypi-artwork
```

I skipped stage4 & stage5, so not sure what else is also missing from `bullseye` channel.

I am very curious about the official os image offered for download on the pi's website, how did they manage to build it without problem? Did they use `buster` instead of `bullseye`? But from the release note, the kernel is 5.10, which should be `bullseye`

free-x, 2021-06-06:
since in bullseye transition from python2 to python3 is completed, only python3-* packages should be used

XECDesign, 2021-06-07:
Official images are Buster, Bullseye is not offered or supported by us right now. Raspberry Pi OS Buster is on 5.10.

free-x, 2021-06-15:
debian plans to release bullseye on 31.7
https://lists.debian.org/debian-release/2021/06/msg00199.html

it would be nice if the packages/repositories of Raspberry were ready too

Regards
free-x

tstachl, 2021-07-11:
With **Bullseye** the format of the `/etc/apt/sources.list` line for the security repository has [changed](https://wiki.debian.org/NewInBullseye). Additionally, I had to remove two packages from the `stage2/01-sys-tweaks/00-packages` list: https://github.com/RPi-Distro/pi-gen/compare/arm64...pilina:arm64-bullseye

I did **not** test `stage3` to `stage5`.

pirateradiohack, 2022-03-30:
I'm not sure to understand, would you care to clear something for me please: I hadn't synced my version of pi-gen for a few months, I have just updated it. And now when I try to build I also get the missing packages errors for wiring pi and others. I did not specify any release, it's just the defaults. So is it Bullseye by default now? Should I specify Buster to make sure all my packages still install correctly as they used to until a few months ago? Thanks a lot for your help.

XECDesign, 2022-03-31:
> So is it Bullseye by default now?

That's right.

> Should I specify Buster to make sure all my packages still install correctly as they used to until a few months ago?

That won't work because pi-gen will still try to install packages which aren't available in Buster. If you want to build Buster images, you should do so from the `buster` branch.

> And now when I try to build I also get the missing packages errors for wiring pi and others. I did not specify any release, it's just the defaults.

I'm a little confused by that, because it would imply that you're building Buster images from the `master` branch. What does your `config` file look like and how are you starting the build?

pirateradiohack, 2022-04-02:
Thanks a lot for the heads up @XECDesign.

Indeed I was trying to build Buster from master.

I have now switched to the Buster branch and see the following message from Github: _This branch is 5 commits ahead, 40 commits behind master. _ Which makes me wonder if the Buster branch is maintained, and what benefit I would have to keep syncing my own pi-gen repo with this branch in the future.

XECDesign, 2022-04-04:
> I have now switched to the Buster branch and see the following message from Github: _This branch is 5 commits ahead, 40 commits behind master_. Which makes me wonder if the Buster branch is maintained, and what benefit I would have to keep syncing my own pi-gen repo with this branch in the future.

Not sure what you mean there. That's not a message from github, it just looks like your buster branch is set to track the master branch (which shouldn't be the case).

The buster branch is currently used to build the 'legacy' images, so yes, it's maintained.

> It seems unlikely that those packages are actually missing. Did I do something wrong?

No, that looks right.

http://wiringpi.com/wiringpi-deprecated/

https://lists.debian.org/debian-python/2019/07/msg00080.html

pirateradiohack, 2022-04-04:
Thanks a lot for clarifying things. I always get confused with those versions upgrades, since the very first versions of Debian.

XECDesign, 2022-04-04:
No worries. Going to close the issue since it doesn't seem relevant anymore.

closed 2022-04-04 by XECDesign