RPi-Distro/pi-gen #508: Problem to run commands with sudo in on_chroot

slmarcos, 2021-04-25:
Hello, I'm creating a custom script in order to install docker and docker compose, but when i try run any command with prefix sudo I receive this error:

```
+ sudo true
Unknown host QEMU_IFLA type: 50
Unknown host QEMU_IFLA type: 51
Unknown host QEMU_IFLA type: 50
Unknown host QEMU_IFLA type: 51
sudo: unable to resolve host a98285367776: No address associated with hostname
```

I'm running the process using the build-docker.sh

tstachl, 2021-07-11:
Add a custom stage to your build with the following files:

`docker/00-install/files/docker.list`
```
deb [arch=arm64 signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/debian RELEASE stable
```

`docker/00-install/00-pckages`
```
apt-transport-https
ca-certificates
curl
gnupg
```

`docker/00-install/01-run.sh` < make sure to `chmod a+x` this file.
```
#!/bin/bash -e

install -m 644 files/docker.list "${ROOTFS_DIR}/etc/apt/sources.list.d/"
sed -i "s/RELEASE/${RELEASE}/g" "${ROOTFS_DIR}/etc/apt/sources.list.d/docker.list"

on_chroot << EOF
curl -fsSL https://download.docker.com/linux/debian/gpg | gpg --dearmor -o /usr/share/keyrings/docker-archive-keyring.gpg

apt-get update
apt-get install -y docker-ce docker-ce-cli containerd.io

adduser ${FIRST_USER_NAME} docker
systemctl enable docker.service
systemctl enable containerd.service
EOF
```

This will install docker for you, you'll have to modify the script to also install docker-compose.

starbasessd, 2021-11-14:
Many thanks @tstachl, this set of instructions / hints helped me resolve a number of issues I was having with python2 and 3 and getting some python modules installed.

XECDesign, 2022-01-20:
IIRC, the original issue is not fatal. Normally sudo would be able to tell from /etc/hosts where the hostname points to.

https://github.com/RPi-Distro/pi-gen/blob/288b21fc27e128ea6b330777aca68e0061ebf4fe/stage1/02-net-tweaks/00-run.sh

If for whatever reason, it can't, it will give that warning, but still work.

I'm guessing this is a docker-specific issue and doesn't happen otherwise? qemu could also be at fault, depending on the version used. Once there's a known-good version of qemu out there, I can make pi-gen fetch it and use it automatically.