RPi-Distro/pi-gen #512: bsdtar: Package bsdtar is not available.

myxxmikeyxx, 2021-05-25:
Package bsdtar is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libarchive-tools:i386 libarchive-tools

This is what I am getting from Ubuntu when trying to install the dependencies. 
Will libarchive-tools work instead?

XECDesign, 2021-05-26:
Yes, the dependencies should be updated to use libarchive-tools, since that's the package bsdtar comes from. I believe bsdtar was a transitional package at some point.

myxxmikeyxx, 2021-05-26:
Then the readme.md should be changed to
`apt-get install coreutils quilt parted qemu-user-static debootstrap zerofree zip \
dosfstools libarchive-tools libcap2-bin grep rsync xz-utils file git curl bc \
qemu-utils kpartx`

closed 2021-05-26 by myxxmikeyxx

miccies3, 2021-06-23:
Hey, there was an update #526. I think the issue can be closed.