RPi-Distro/pi-gen #513: Custom pi-gen build gives unmet dependencies failure for Python

lvoid, 2021-05-25:
I am trying to build a custom Raspberry Pi OS image for a touch screen Kivy/KivyMD app that has QR code scanning features (hence uses OpenCV and pyzbar, along with all dependencies) and Kali Linux tools.

I've customized a few steps in the build stages as follows:

 - **Stage0** - Added Kali Linux repo `deb https://http.kali.org/kali kali-rolling main contrib non-free` to `stage0/00-configure-apt/files/sources.list`
- **Stage2** - Edited `/pi-gen/stage2/01-sys-tweaks/00-packages` to include a number of dependencies I need:
```
python3-setuptools
git-core
python3-dev
pkg-config
libgl1-mesa-dev
libgles2-mesa-dev
libgstreamer1.0-dev
gstreamer1.0-plugins-{bad,base,good,ugly}
gstreamer1.0-{omx,alsa}
libmtdev-dev
xclip
xsel
libjpeg-dev
libsdl2-dev
libsdl2-image-dev
libsdl2-mixer-dev
libsdl2-ttf-dev
```
- **Stage3** - Edited `/pi-gen/stage3/00-install-packages/01-run.sh` to install additional dependencies by appending these lines to the script:
```
apt-get update -y
apt-get install libzbar0=0.22-1 --allow-downgrades -y
apt-get install python-zbar --allow-downgrades -y
apt-get install libhdf5-dev --allow-downgrades -y
apt-get install libhdf5-serial-dev --allow-downgrades -y
apt-get install libatlas-base-dev --allow-downgrades -y
apt-get install libjasper-dev --allow-downgrades -y
apt-get install libqt5gui5 --allow-downgrades -y
```

The build process fails on **Stage4** when the build process is installing packages from `pi-gen/stage4/00-install-packages/00-packages`.

It fails with the following information:

```
The following packages have unmet dependencies:
 libpython2.7-minimal : Breaks: python (< 2.7.18)
 python : PreDepends: python-minimal (= 2.7.16-1) but it is not installable
          Depends: libpython-stdlib (= 2.7.16-1) but it is not installable
          Depends: python2 (= 2.7.16-1) but 2.7.18-2 is to be installed
 python-oauthlib : Depends: python-cryptography but it is not installable
 python-pip : Depends: python-pip-whl (= 18.1-5+rpt1) but 20.3.4-2 is to be installed
              Recommends: python-all-dev (>= 2.6) but it is not going to be installed
              Recommends: python-wheel but it is not going to be installed
 python2.7 : Breaks: python (< 2.7.18)
 python2.7-minimal : Breaks: python (< 2.7.18)
 python3-chardet : Breaks: python-chardet (< 3.0.4-6) but 3.0.4-3 is to be installed
 python3-pip : Breaks: python-pip but 18.1-5+rpt1 is to be installed
 python3-rtimulib : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
 python3-spidev : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Packages (contrib/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Packages (contrib/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Translations (contrib/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Packages (non-free/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Packages (non-free/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
W: Target Translations (non-free/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:2 and /etc/apt/sources.list:3
```

This is obviously a problem with dependencies, especially Python, but I am not quite sure how to fix it or how the dependencies I added to the pi-gen steps have caused this error.

Does anyone know how to fix this dependency issue?  Are any packages I installed causing the error?

XECDesign, 2021-05-26:
You can't mix repos which aren't intended to work together like that.

https://wiki.debian.org/DontBreakDebian#Don.27t_make_a_FrankenDebian

lvoid, 2021-05-26:
@XECDesign Thank you, I was actually able to install the tools I needed without that repo.

I got to the stage of `export-noobs` and it froze after it brought up my file explorer and said `Could not find "/home/joe/pi-gen/2021-05-25-test/tmpimage/boot"`

Any idea where I could go from here?  The last line to be generated by pi-gen in my terminal is:

`'/home/joe/pi-gen/work/2021-05-26-test/export-noobs/rootfs/etc/systemd/system/multi-user.target.wants/apply_noobs_os_config.service' -> '/lib/systemd/system/apply_noobs_os_config.service'`

XECDesign, 2021-05-27:
Sorry, not much to go on there.

closed 2021-05-27 by lvoid