RPi-Distro/pi-gen #515: Read error, sector 0

myxxmikeyxx, 2021-05-26:
I am running into this error and I am not sure what is wrong.

`[11:01:27] Begin /home/install17/pi-gen/stage4
Creating backing image: image-stage4.qcow2 <- /home/install17/pi-gen/work/2021-05-26-Raspbian/image-stage3.qcow2
Formatting '/home/install17/pi-gen/work/2021-05-26-Raspbian/image-stage4.qcow2', fmt=qcow2 size=12884901888 backing_file=/home/install17/pi-gen/work/2021-05-26-Raspbian/image-stage3.qcow2 cluster_size=65536 lazy_refcounts=off refcount_bits=16
read error, sector 0
llseek error
llseek error
llseek error
mount: /home/install17/pi-gen/work/2021-05-26-Raspbian/stage4/rootfs: special device /dev/mapper/nbd0p2 does not exist.
[11:01:28] Unloading image
`
The build.log file is bellow:
`[10:18:05] Begin /home/install17/pi-gen
[10:18:05] Begin /home/install17/pi-gen/stage0
[10:18:11] Begin /home/install17/pi-gen/stage0/prerun.sh
[10:29:24] End /home/install17/pi-gen/stage0/prerun.sh
[10:29:24] Begin /home/install17/pi-gen/stage0/00-configure-apt
[10:29:24] Begin /home/install17/pi-gen/stage0/00-configure-apt/00-run.sh
[10:30:03] End /home/install17/pi-gen/stage0/00-configure-apt/00-run.sh
[10:30:03] End /home/install17/pi-gen/stage0/00-configure-apt
[10:30:03] Begin /home/install17/pi-gen/stage0/01-locale
[10:30:03] Begin /home/install17/pi-gen/stage0/01-locale/00-debconf
[10:30:04] End /home/install17/pi-gen/stage0/01-locale/00-debconf
[10:30:04] Begin /home/install17/pi-gen/stage0/01-locale/00-packages
[10:30:51] End /home/install17/pi-gen/stage0/01-locale/00-packages
[10:30:51] End /home/install17/pi-gen/stage0/01-locale
[10:30:51] Begin /home/install17/pi-gen/stage0/02-firmware
[10:30:51] Begin /home/install17/pi-gen/stage0/02-firmware/01-packages
[10:33:50] End /home/install17/pi-gen/stage0/02-firmware/01-packages
[10:33:50] End /home/install17/pi-gen/stage0/02-firmware
[10:33:50] Begin /home/install17/pi-gen/stage0/files
[10:33:50] End /home/install17/pi-gen/stage0/files
[10:33:53] End /home/install17/pi-gen/stage0
[10:33:53] Begin /home/install17/pi-gen/stage1
[10:33:55] Begin /home/install17/pi-gen/stage1/prerun.sh
[10:33:55] End /home/install17/pi-gen/stage1/prerun.sh
[10:33:55] Begin /home/install17/pi-gen/stage1/00-boot-files
[10:33:55] Begin /home/install17/pi-gen/stage1/00-boot-files/00-run.sh
[10:33:55] End /home/install17/pi-gen/stage1/00-boot-files/00-run.sh
[10:33:55] End /home/install17/pi-gen/stage1/00-boot-files
[10:33:55] Begin /home/install17/pi-gen/stage1/01-sys-tweaks
[10:33:55] Begin /home/install17/pi-gen/stage1/01-sys-tweaks/00-patches
[10:33:56] End /home/install17/pi-gen/stage1/01-sys-tweaks/00-patches
[10:33:56] Begin /home/install17/pi-gen/stage1/01-sys-tweaks/00-run.sh
[10:33:59] End /home/install17/pi-gen/stage1/01-sys-tweaks/00-run.sh
[10:33:59] End /home/install17/pi-gen/stage1/01-sys-tweaks
[10:33:59] Begin /home/install17/pi-gen/stage1/02-net-tweaks
[10:33:59] Begin /home/install17/pi-gen/stage1/02-net-tweaks/00-packages
[10:34:09] End /home/install17/pi-gen/stage1/02-net-tweaks/00-packages
[10:34:09] Begin /home/install17/pi-gen/stage1/02-net-tweaks/00-run.sh
[10:34:09] End /home/install17/pi-gen/stage1/02-net-tweaks/00-run.sh
[10:34:09] End /home/install17/pi-gen/stage1/02-net-tweaks
[10:34:09] Begin /home/install17/pi-gen/stage1/03-install-packages
[10:34:09] Begin /home/install17/pi-gen/stage1/03-install-packages/00-packages
[10:35:21] End /home/install17/pi-gen/stage1/03-install-packages/00-packages
[10:35:21] End /home/install17/pi-gen/stage1/03-install-packages
[10:35:24] End /home/install17/pi-gen/stage1
[10:35:24] Begin /home/install17/pi-gen/stage2
[10:35:25] Begin /home/install17/pi-gen/stage2/prerun.sh
[10:35:25] End /home/install17/pi-gen/stage2/prerun.sh
[10:35:25] Begin /home/install17/pi-gen/stage2/00-copies-and-fills
[10:35:25] Begin /home/install17/pi-gen/stage2/00-copies-and-fills/01-packages
[10:35:42] End /home/install17/pi-gen/stage2/00-copies-and-fills/01-packages
[10:35:42] Begin /home/install17/pi-gen/stage2/00-copies-and-fills/02-run.sh
[10:35:42] End /home/install17/pi-gen/stage2/00-copies-and-fills/02-run.sh
[10:35:42] End /home/install17/pi-gen/stage2/00-copies-and-fills
[10:35:42] Begin /home/install17/pi-gen/stage2/01-sys-tweaks
[10:35:42] Begin /home/install17/pi-gen/stage2/01-sys-tweaks/00-debconf
[10:35:44] End /home/install17/pi-gen/stage2/01-sys-tweaks/00-debconf
[10:35:44] Begin /home/install17/pi-gen/stage2/01-sys-tweaks/00-packages-nr
[10:36:07] End /home/install17/pi-gen/stage2/01-sys-tweaks/00-packages-nr
[10:36:07] Begin /home/install17/pi-gen/stage2/01-sys-tweaks/00-packages
[10:49:07] End /home/install17/pi-gen/stage2/01-sys-tweaks/00-packages
[10:49:07] Begin /home/install17/pi-gen/stage2/01-sys-tweaks/00-patches
[10:49:07] End /home/install17/pi-gen/stage2/01-sys-tweaks/00-patches
[10:49:07] Begin /home/install17/pi-gen/stage2/01-sys-tweaks/01-run.sh
[10:49:26] End /home/install17/pi-gen/stage2/01-sys-tweaks/01-run.sh
[10:49:26] End /home/install17/pi-gen/stage2/01-sys-tweaks
[10:49:26] Begin /home/install17/pi-gen/stage2/02-net-tweaks
[10:49:26] Begin /home/install17/pi-gen/stage2/02-net-tweaks/00-packages
[10:50:17] End /home/install17/pi-gen/stage2/02-net-tweaks/00-packages
[10:50:17] Begin /home/install17/pi-gen/stage2/02-net-tweaks/01-run.sh
[10:50:17] End /home/install17/pi-gen/stage2/02-net-tweaks/01-run.sh
[10:50:17] End /home/install17/pi-gen/stage2/02-net-tweaks
[10:50:17] Begin /home/install17/pi-gen/stage2/03-accept-mathematica-eula
[10:50:17] Begin /home/install17/pi-gen/stage2/03-accept-mathematica-eula/00-debconf
[10:50:18] End /home/install17/pi-gen/stage2/03-accept-mathematica-eula/00-debconf
[10:50:18] End /home/install17/pi-gen/stage2/03-accept-mathematica-eula
[10:50:18] Begin /home/install17/pi-gen/stage2/03-set-timezone
[10:50:18] Begin /home/install17/pi-gen/stage2/03-set-timezone/02-run.sh
[10:50:21] End /home/install17/pi-gen/stage2/03-set-timezone/02-run.sh
[10:50:21] End /home/install17/pi-gen/stage2/03-set-timezone
[10:50:23] End /home/install17/pi-gen/stage2
[10:50:23] Begin /home/install17/pi-gen/stage3
[10:50:27] Begin /home/install17/pi-gen/stage3/prerun.sh
[10:50:27] End /home/install17/pi-gen/stage3/prerun.sh
[10:50:27] Begin /home/install17/pi-gen/stage3/00-install-packages
[10:50:27] Begin /home/install17/pi-gen/stage3/00-install-packages/00-debconf
[10:50:30] End /home/install17/pi-gen/stage3/00-install-packages/00-debconf
[10:50:30] Begin /home/install17/pi-gen/stage3/00-install-packages/00-packages-nr
[10:56:09] End /home/install17/pi-gen/stage3/00-install-packages/00-packages-nr
[10:56:09] Begin /home/install17/pi-gen/stage3/00-install-packages/00-packages
[11:01:21] End /home/install17/pi-gen/stage3/00-install-packages/00-packages
[11:01:21] Begin /home/install17/pi-gen/stage3/00-install-packages/01-run.sh
[11:01:22] End /home/install17/pi-gen/stage3/00-install-packages/01-run.sh
[11:01:22] End /home/install17/pi-gen/stage3/00-install-packages
[11:01:22] Begin /home/install17/pi-gen/stage3/01-tweaks
[11:01:22] Begin /home/install17/pi-gen/stage3/01-tweaks/00-run.sh
[11:01:22] End /home/install17/pi-gen/stage3/01-tweaks/00-run.sh
[11:01:22] End /home/install17/pi-gen/stage3/01-tweaks
[11:01:27] End /home/install17/pi-gen/stage3
[11:01:27] Begin /home/install17/pi-gen/stage4
[11:01:28] Unloading image`

myxxmikeyxx, 2021-05-27:
I am using Ubuntu 20.04 LTS in a virtual machine on virtual box. I don't think that should matter but I wanted to point this out incase it helps to find the problem.

XECDesign, 2021-05-27:
@pandel may be able to help with that one.

In the meanwhile, you could try building with `USE_QCOW2=0`

pandel, 2021-05-27:
@myxxmikeyxx 
I have no idea what's happening here. You are too far into the build to just say, that your NBDs simply don't work. If I find time, I'll setup a virtual machine like you did and try to find a reason for this.

myxxmikeyxx, 2021-05-27:
On Ubuntu 16 I am getting this error:
[image]

myxxmikeyxx, 2021-05-27:
root@ubuntu16VM:/home/install17/pi-gen/stage0# ./prerun.sh 
./prerun.sh: line 4: bootstrap: command not found
root@ubuntu16VM:/home/install17/pi-gen/stage0#

pandel, 2021-05-27:
What I don't understand is the device name `/dev/mapper/nbd10p2`. So inside the build process the system is trying to access nbd10, but it is nbd1 actually. I am not sure why this is happening and have never noticed such behaviour in my environment...

pandel, 2021-05-27:
You cannot issue the prerun.sh command outside the build process, as it is executed inside the docker environment...

pandel, 2021-05-27:
OH! Are you trying to use build.sh in a non-ARM environment? That does not work. You must use the Docker build variant then!

pandel, 2021-05-27:
Try using the Docker build variant via build-docker.sh perhaps...

myxxmikeyxx, 2021-05-27:
I have switched over to a raspberry pi 3. I am testing doing the build on it. 
The build seems to be working, slower, so I do not know if it will get hung up on the same spot.
I also downloaded the x86 Raspbian img and I am trying that on the Virtual Box, I just got all the dependences and cloned it.
I will update you on how it goes. If it fails then I will try it back on Ubuntu (both) and try the docker build to see if it works.

myxxmikeyxx, 2021-05-27:
Also thank you @pandel for trying to help. I am not super tech savvy. I really want to get this to work so I can create my own IMG that has different packages pre-installed. My goal is to basically try to make a  [LazyCast](https://github.com/homeworkc/lazycast) img file. That still works with Raspberry Pi Imager. 
I want it to have the necessary packages but still be able to get the Wi-Fi info and then I parse that info from the wpa_supplicant and run a script to take the Wi-Fi name and password and connect to it through network-manager in command line.
`sudo nmcli dev wifi connect 'Wi-Fi Name' password 'my Wi-Fi Password`

myxxmikeyxx, 2021-05-27:
Raspberry pi x86 error:
[image]

Raspberry pi: Still running

pandel, 2021-05-28:
@myxxmikeyxx
Strange, as the build succeeded itself, only image conversion at the end failed. I know that this "failed to setup loop device" error was something, that could happen in an older Docker release on my Mac, too, and I also noticed it in a Debian ARM64 VM on my Proxmox system. But it was related to timing issues in combination with read/write cycles of the Proxmox hypervisor and solved itself via a Proxmox kernel update.
I am really sorry, but do not have any idea, how I could help you without a deep dive into your system by hand, which isn't possible, obviously...

rianadon, 2021-06-01:
I get the same error message. Ubuntu 20.04, on vmware esxi. I'm building with `./build-docker`:

```
[01:25:22] End /pi-gen/stage1/03-install-packages/00-packages
[01:25:22] End /pi-gen/stage1/03-install-packages
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs: 10.9 GiB (11649982464 bytes) trimmed
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs/sys
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs/proc
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs/dev/pts
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs/dev
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs/boot
/pi-gen/work/2021-06-01-Raspbian/stage1/rootfs
/dev/nbd0 disconnected
[01:25:22] End /pi-gen/stage1
[01:25:22] Begin /pi-gen/stage2
Creating backing image: image-stage2.qcow2 <- /pi-gen/work/2021-06-01-Raspbian/image-stage1.qcow2
Formatting '/pi-gen/work/2021-06-01-Raspbian/image-stage2.qcow2', fmt=qcow2 size=12884901888 backing_file=/pi-gen/work/2021-06-01-Raspbian/image-stage1.qcow2 cluster_size=65536 lazy_refcounts=off refcount_bits=16
read error, sector 0
llseek error
llseek error
llseek error
mount: /pi-gen/work/2021-06-01-Raspbian/stage2/rootfs: special device /dev/mapper/nbd0p2 does not exist.
[01:25:22] Unloading image
```

Building with `USE_QCOW2=0` works, so that's what I'm doing for now.

pandel, 2021-06-02:
@rianadon
Thanks for your information. It must have something to do with 20.04. Unfortunately, I am not able to test this atm, as I am still moving and busy. I try to get back to this as soon as possible...

rianadon, 2021-06-02:
Thanks for looking into this @pandel! For me at least, this isn't a blocker (qcow2 disabled fixes the read error sector 0 problem and successfully builds).

I wonder if it's running in a virtual machine that could be the problem as well. I was running inside docker, which as far as I understand, would eliminate ubuntu quirks. If it would help I also have an Arch VM I could try building inside to test the VM theory.