RPi-Distro/pi-gen #518: `dbus` error when trying to use `bluetoothctl` in `chroot`

pirateradiohack, 2021-06-03:
I need to setup bluetooth. Part of the configuration requires the use of `bluetoothctl`. I tried to use `bluetoothctl` in a `run-chroot.sh` but `dbus` crashed during the build:

`dbus[30266]: arguments to dbus_connection_get_object_path_data() were incorrect, assertion "connection != NULL" failed in file ../../../dbus/dbus-connection.c line 5905.`

I guess that's to be expected since there is no bluetooth adapter during build. 

It's probably a bit off topic, but any clue as to how to setup bluetooth things such as `power on`, `discoverable on` and `pairable on` baked into the image would be great. 

Thanks for your help.

XECDesign, 2021-06-04:
Sorry, no idea. I'd try to figure out where the data ends up being stored and put it there directly rather than through bluetoothctl.

pirateradiohack, 2021-06-04:
Thanks for your input.
Yes that's what I want to do. I need to figure out where that gets stored, I'm not very familiar with bluetoothctl.

closed 2021-06-04 by pirateradiohack