RPi-Distro/pi-gen #524: debootstrap.log is removed on failure

voltagex, 2021-06-17:
See https://builds.sr.ht/~voltagex/job/527248, specifically

```
W: Failure trying to run: chroot "/home/build/pi-gen/work/2021-06-17-test/stage0/rootfs" /bin/true
W: See /home/build/pi-gen/work/2021-06-17-test/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/home/build/pi-gen/work/2021-06-17-test/stage0/rootfs/debootstrap': Directory not empty
```

```
build@build:~/pi-gen$ less /home/build/pi-gen/work/2021-06-17-test/stage0/rootfs/build.log
/home/build/pi-gen/work/2021-06-17-test/stage0/rootfs/build.log: No such file or directory
```

So then I can't troubleshoot easily because the log has been removed

voltagex, 2021-06-17:
It's this, but I don't know why. 
https://github.com/RPi-Distro/pi-gen/issues/338

voltagex, 2021-06-17:
seems to be a transient sr.ht failure.

XECDesign, 2021-06-17:
It means debootstrap failed, so I'd check the log file it generates.

XECDesign, 2021-06-17:
Sorry, missed the part where you mentioned the log file already.

I don't see pi-gen removing the log file. It tries to remove the directory and fails if it's not empty, which is the intended behaviour. On succesfull bootstrap, that directory should be empty.

Without being able to reproduce the issue myself, I can't really help.

voltagex, 2021-06-21:
Right, the build system was unmounting things before I could check what was going on.

The underlying error appears to be

```
build@build:~$ sudo chroot "/home/build/pi-gen/work/2021-06-21-test/stage0/rootfs" /bin/true
chroot: failed to run command '/bin/true': Exec format error
```

Which I thought was handled by binfmt_misc and qemu-static.

voltagex, 2021-06-21:
Sorry, I'd missed the `binfmt-support` package requirement. I still think there's a potential bug with the debootstrap.log being "hidden" in the now-unmounted filesystem.

XECDesign, 2021-06-21:
What do you think would be an intuitive way to cope with that?

I didn't originally include qcow2 support was because it adds an extra layer of things that can go wrong and makes troubleshooting more difficult. Should the default be to not use qcow? Should pi-gen 'tail' the output of debootstrap.log if it's present?

voltagex, 2021-06-21:
@XECDesign IMO qcow2 is the right thing to use, what about grabbing debootstrap.log and copying it out of the work path before unmounting?

XECDesign, 2021-06-22:
Something along these lines?

https://github.com/RPi-Distro/pi-gen/commit/f9d44edb72e9a153c997266189c79cb1f96b69f3

closed 2021-07-03 by voltagex

nullromo, 2024-07-25:
I'm having the same error

`chroot: failed to run command ‘/bin/true’: Exec format error`

showing up in debootstrap.log. I don't understand what this means. @voltagex any info on how you solved this?

I could open a new issue if that's better.

voltagex, 2024-07-26:
@nullromo no, this was 3 years ago. Exec format error means that the binary is the wrong type for your machine, i.e. arm binary trying to run on an x86_64 system.

nullromo, 2024-07-26:
Thanks. I think I got confused about something, and I was running the wrong version of my code for a while there. I opened a [new issue](https://github.com/RPi-Distro/pi-gen/issues/782) instead.