RPi-Distro/pi-gen #525: Set BOOT_UART=1 with pi-gen

capiman, 2021-06-18:
I found the following documentation about enabling more debug during booting:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/README.md

It tells me to query

$ strings bootcode.bin | grep BOOT_UART
BOOT_UART=0

To enable UART from bootcode.bin use:

sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin

How/where can I do this inside pi-gen? In which stage bootcode.bin is handled?

I don't want to mount the final generated image. Or install the image and do the replacement.
I want to add it directly inside the image generator, as additional step in one of the stages.

XECDesign, 2021-06-22:
I'd probably create `export-image/04-finalise/00-run.sh` with the following content:

```
#!/bin/bash -e
sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" ${ROOTFS_DIR}/boot/bootcode.bin
```

capiman, 2021-06-22:
Thanks for your answer. I just have done it successfully this afternoon similar in the way you suggested.
But I had put it into stage0/50-enable-uart-in-bootcode

XECDesign, 2021-06-22:
That'll work as long as you don't re-use stages for too long. If the firmware gets upgraded during export, you'll lose that change, but that's very unlikely.

closed 2021-06-22 by XECDesign

capiman, 2021-06-22:
Thanks for the hint. I will then move it to export-image/04-finalise/00-run.sh