RPi-Distro/pi-gen #534: Building an image only for RPI-Zero

skast96, 2021-07-21:
Hey Guys

short question I have got some compiled cython packages that I use on my Pi-Zero these are compiled with armv6 and are not supported by the pi-gen runner, which is using the armv7 set. Is there a possible fix to run the pi-gen only for armv6? Or do you guys have any other solutions?

marcone, 2021-08-31:
Is there any particular reason you want the image to not support the other Pi versions at all? If it's to save space in the image, what I've done in the past is to add a custom final stage with a "00-run.sh" that removes the modules I don't need, e.g:
```
on_chroot << EOF
rm -rf /lib/modules/*-v7+
rm -rf /lib/modules/*-v8+
EOF
```
(the above leaves only the modules needed for Pi0 and 32-bit Pi4, saving about 140 MB).

skast96, 2021-09-07:
> Is there any particular reason you want the image to not support the other Pi versions at all? If it's to save space in the image, what I've done in the past is to add a custom final stage with a [...]

It's because we had software that was only compiled for the zero and should only run on a bunch of zeros we had, before switching to a CM4.

XECDesign, 2021-09-07:
Raspberry Pi OS images are built with armv6-only pis in mind, so I'm not sure I understand what the issue is.

skast96, 2021-09-17:
When building the image with the docker container it is saying armv7 when running "uname -a". Therefore I get an error while the docker container is trying to install my armv6 built software.

XECDesign, 2021-09-20:
I'm not familiar enough with Docker or cython to be able to help here. I'd expect there to be a flag you could pass to specify that you want an armv6 container. Alternatively, there are ways to spoof uname output. This doesn't sound like a pi-gen issue.

closed 2021-09-20 by XECDesign