RPi-Distro/pi-gen #537: root size is incorrectly determined

marcone, 2021-08-15:
`scripts/qcow2_handling` uses `resize2fs -P` to determine the minimum size of the root fs, then adds 1 MB to that.
However it appears that `resize2fs` is wildly wrong about the size. In my stage2 build it returns 420833 blocks, and indeed it ends up creating an image with a 1.6 GB root partition:
```
Device                        Boot  Start     End Sectors  Size Id Type
2021-08-15-teslausb-lite.img1 *      8192  520191  512000  250M  c W95 FAT32 (LBA)
2021-08-15-teslausb-lite.img2      520192 3821040 3300849  1.6G 83 Linux
```
However if I actually mount that partition I see that there is still about 340 MB of free space in it:
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/loop0p2    1.5G  1.1G  340M  77% /home/marcone/pi-gen/deploy/mnt
```
Perhaps it would be better to use `du` to determine the size of the file system, like `export-image/prerun.sh` does? Except: that script uses `du --apparent-size ...` which also gives wildly incorrect results for me: a regular `du` shows 1.1 GB in use as it should, however `du --apparent-size` shows over 120 TB in use, which is clearly incorrect.

Finally, it would be great if the padding was consistent between the qcow2 and the non-qcow2 paths (it is currently 1 MB for one and 20 percent plus 200 MB for the other) and was configurable as a config option. This is important for use cases where the root partition will not be auto-expanded to fill the remainder of the disk on first boot.

XECDesign, 2021-08-31:
Whatever approach we use, I agree that the output should be identical regardless of whether you use qcow2 or not.

I'm inclined to trust resize2fs and suspect there may be a reason why it leaves so much free space. I'd like to just create 8GB images and shrink them with `resize2fs -M` then expand again to add the appropriate amount of padding, which should probably be a static number.

Any reason that would be a bad approach?

marcone, 2021-08-31:
If by "static number" you mean a fixed value that can be overridden in the config file, then that seems like a good approach. I wouldn't just hardcode that number in the build scripts themselves like it is now.

XECDesign, 2021-08-31:
Alright, added to todo list.

wscheicher, 2021-09-12:
The biggest problem with `du --apparent-size` as it is used currently: it includes mounted /proc which gives 128T on my system.
So maybe -x / --one-file-system should be used?
Or at least /proc be excluded?

wscheicher, 2021-09-16:
Ok, seems that /proc should have been unmounted at time of du, but it looks like that fails if you work with symlinks.
After specifying WORK_DIR to a path without symlinks things worked again.

commit f1c166a2 (2024-04-29): export-image: run du with -x to avoid crossing filesystem boundaries

Fixes #537
closed 2024-04-29 by XECDesign