RPi-Distro/pi-gen #543: Reopen: Failure while configuring base packages

woohsian, 2021-09-06:
Just going to copy the problem from here https://github.com/RPi-Distro/pi-gen/issues/446:

Basically I notice that this step will not work for the whole day on Saturday / Sunday and then without further intervention rerunning works at night. I suspect someone is turning their servers off during this time. Is there a way to make this more robust so it can work at any day / time?

```
I: Configuring gpg-wks-client...
I: Configuring man-db...
I: Configuring gnupg...
I: Configuring raspbian-archive-keyring...
I: Configuring systemd...
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)
I: Configuring raspbian-archive-keyring...
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package raspbian-archive-keyring is at fault)
```
(the above then exits)

Actual `debootstrap.log`
```
Setting up raspbian-archive-keyring (20120528.2) ...
Warning: apt-key should not be used in scripts (called from postinst maintainerscript of the package raspbian-archive-keyring)
gpg: can't connect to the agent: IPC connect call failed
dpkg: error processing package raspbian-archive-keyring (--configure):
installed raspbian-archive-keyring package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of apt:
apt depends on raspbian-archive-keyring; however:
  Package raspbian-archive-keyring is not configured yet.

dpkg: error processing package apt (--configure):
dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of apt-utils:
apt-utils depends on apt (= 1.8.2.3); however:
  Package apt is not configured yet.

dpkg: error processing package apt-utils (--configure):
dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of tasksel:
tasksel depends on apt; however:
  Package apt is not configured yet.

dpkg: error processing package tasksel (--configure):
dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of tasksel-data:
tasksel-data depends on tasksel (= 3.53); however:
  Package tasksel is not configured yet.

dpkg: error processing package tasksel-data (--configure):
dependency problems - leaving unconfigured
Errors were encountered while processing:
raspbian-archive-keyring
apt
apt-utils
tasksel
tasksel-data
```

XECDesign, 2021-09-08:
```
gpg: can't connect to the agent: IPC connect call failed
```

I haven't seen that before, but it looks like a local issue.

What's your setup? Are you building natively on arm or not? Are you using Docker or not?

I suspect it's Docker breaking gnupg's communication with its daemon.

woohsian, 2021-09-08:
My bad - I might have misdiagnosed as an external issue.

I am using `build-docker.sh` on Ubuntu 20.04 Desktop and just `ENABLE_SSH=1` setting

Any tips on how to stop the breaking of the gnupg's communication with the daemon? The build seems to fail like 10 times in a row then pass the 11th time

XECDesign, 2021-09-08:
Not sure if it's a docker issue or a qemu issue there. Right now qemu is the cause of 90% of all the woes with pi-gen, but I haven't seen that one in particular. I personally don't use Docker, so I can't say if that's where the problem lies, but if it is, unfortunately I wouldn't be of any help.

In the issue you linked, the problem was with running multiple builds at once. Is there anything else running on that system that might be interfering?

I really have no idea.

woohsian, 2021-09-08:
Thanks for trying to help. 

I actually read the issue the same as you (multiple builds). However now that I re-read the issue again it might be related to the fact I have other containers running (they are not pi-gen containers, but just other Docker containers doing other work). Perhaps Docker is not good with handling multiple containers in general in relation to gpg

I will give that a go and will post here if I find further issues.

closed 2021-09-08 by woohsian