RPi-Distro/pi-gen #546: APT repo seems to have mixed releases: InRelease' changed its 'Suite' value from 'stable' to 'oldstable'

guysoft, 2021-09-12:
Hey,
Apt-get update gets the following result:

```
+ apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.7 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
```

Here is a talk on the rpi forms:
https://www.raspberrypi.org/forums/viewtopic.php?t=318302

CustomPiOS/OctoPi Issue:
https://github.com/guysoft/CustomPiOS/issues/143

guysoft, 2021-09-16:
Also, related: https://github.com/KevinOConnor/klipper/pull/4635

XECDesign, 2021-11-10:
For more context: https://github.com/RPi-Distro/repo/issues/258

closed 2021-11-10 by XECDesign

guysoft, 2021-11-10:
Just to note: I think this still happens in Buster. Which might cause issues in Pis that have not updated.
So if you are reading this the only known workaround is to upgrade. or run:
```
apt-get update --allow-releaseinfo-change
```