RPi-Distro/pi-gen #547: Unloading image

cdupont, 2021-09-15:
I'm trying to make a custom RPI image.
However after a few compilations, I get stuck with this:
```
$ sudo ./build.sh
[14:57:44] Begin /home/cdupont/workspace/pi-gen
[14:57:44] Begin /home/cdupont/workspace/pi-gen/stage0
[14:57:44] End /home/cdupont/workspace/pi-gen/stage0
[14:57:44] Begin /home/cdupont/workspace/pi-gen/stage1
[14:57:44] End /home/cdupont/workspace/pi-gen/stage1
[14:57:44] Begin /home/cdupont/workspace/pi-gen/stage2
[14:57:44] End /home/cdupont/workspace/pi-gen/stage2
[14:57:44] Begin /home/cdupont/workspace/pi-gen/stage3-wazigate
[14:57:44] Unloading image
```
The build stops immediately after printing this message.
`stage3-wazigate` is my custom stage. But it won't be compiled, as the compilation stops.
Thanks for the help!

cdupont, 2021-09-15:
It seems that it is failing because it is missing the file `pi-gen/work/2021-09-15-Raspbian/image-stage2.qcow2` when entering stage 3.
How is it working? This file is used by each stage from the previous stage?

cdupont, 2021-09-16:
I noticed that the work folders are named e.g.  `work/2021-09-15-Raspbian/`.
So a date is inserted before my IMG_NAME... 
So if I build images the same day, it will find the previous qcow2 file. 
However, the next day the build script will not find the qcow2 file and fail as described.

Why is a date inserted in the work folder? What is the expected behavior?

XECDesign, 2021-09-20:
I'd like to think I had a good reason for including the date in the work directory, but I can't think of any right now.

If the changed date is the cause of the issue you're seeing, I can take that out.

cdupont, 2021-09-20:
Hi @XECDesign, yes the date seems to be the problem.
Maybe it can be optional.

XECDesign, 2021-09-21:
Should be resolved by https://github.com/RPi-Distro/pi-gen/commit/2f12cf985a88977957e2743601246d2a07aeb660 then

cdupont, 2021-09-21:
Thank you so much! It seems to work.

XECDesign, 2021-09-21:
Great, thanks for finding the issue!

closed 2021-09-21 by XECDesign