RPi-Distro/pi-gen #549: Failed to fetch ... Error reading from server - read (32: Broken pipe)

beta-tester, 2021-09-22:
i tried pi-gen several times but never reached the end, because the process breaks randomly at fetching packages.
it is not a specific package that can't be fetched, it is random at random packages at random stages.
so i think it is a network related issue.

i can't see any retry messages...

is there an option to retry fetching a package or does it break just after the very first try?

```
Fetched 209 MB in 2min 52s (1214 kB/s)                                                                                                                                                                                                       
E: Failed to fetch http://../pub/Mirrors/archive.raspbian.org/raspbian/pool/main/o/openexr/libopenexr25_2.5.4-2_armhf.deb  Error reading from server - read (32: Broken pipe) [IP: ... 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
[23:12:36] Unloading image
```

when the pi-gen stoped, is there a way to continue installing from the last stoped position?

XECDesign, 2021-09-22:
I think it already does: https://github.com/RPi-Distro/pi-gen/commit/38d22e976a2d6ad2ccabc2704c966f586550122c

We use apt-cacher-ng and point pi-gen to it through `APT_PROXY`, which is great for these situations.

> when the pi-gen stoped, is there a way to continue installing from the last stoped position?

If you skip the stages which have completed, you can continue from where it has failed. I'd recommend using CLEAN=1 to restart the failed stage from scratch. CLEAN=0 would re-run it with the existing files, but I believe that doesn't work too well right now.

beta-tester, 2021-09-22:
yes, you are right. in the build script there is `APT::Acquire::Retries=3` and even putting it to 10 make it not better.
but now i think it is `RELEASE=bullseye` related. i tried it with `RELEASE=buster` everything runs smooth.

so i am maybe too early trying building a bullseye image.