RPi-Distro/pi-gen #550: /dev/mapper/nbd10p1: No such file or directory

1000283, 2021-10-01:
Host: Ubuntu Xenial x64 VM.

I'm trying to build a full image (no SKIPs anywhere) just to test and failing right at stage0.
Using release `2021-05-07-raspbian-buster` and `build.sh`. Same issue on `master`. 

```
$ uname -a
Linux myuser-VirtualBox 4.15.0-129-generic #132~16.04.1-Ubuntu SMP Wed Dec 16 06:46:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
```
$ cat raspbian-full.cfg 
IMG_NAME="Raspbian"
RELEASE="buster"
DEPLOY_ZIP=0
LOCALE_DEFAULT="pt_PT.UTF-8"
KEYBOARD_KEYMAP="pt"
IKEYBOARD_LAYOUT="Portuguese"
TIMEZONE_DEFAULT="Europe/Lisbon"
FIRST_USER_NAME="myuser"
FIRST_USER_PASS="myuser"
ENABLE_SSH=1
```
```
$ sudo CLEAN=1 ./build.sh -c raspbian-full.cfg 
[10:41:44] Begin /home/myuser/dev/pi-gen
[10:41:44] Begin /home/myuser/dev/pi-gen/stage0
Creating base image: image-stage0.qcow2
Formatting 'image-stage0.qcow2', fmt=qcow2 size=12884901888 encryption=off cluster_size=65536 lazy_refcounts=off refcount_bits=16
Checking that no-one is using this disk right now ... OK

Disk /dev/nbd10: 12 GiB, 12884901888 bytes, 25165824 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0xe2ad8441.
Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 250 MiB.
/dev/nbd10p2: Created a new partition 2 of type 'Linux' and of size 11,8 GiB.
/dev/nbd10p3: 
New situation:

Device       Boot  Start      End  Sectors  Size Id Type
/dev/nbd10p1 *      8192   520191   512000  250M  c W95 FAT32 (LBA)
/dev/nbd10p2      520192 25165823 24645632 11,8G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
mkfs.fat 3.0.28 (2015-05-16)
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
/dev/mapper/nbd10p1: No such file or directory
[10:41:44] Unloading image
```
Not much in the log:
```
$ cat work/2021-10-01-Raspbian/build.log 
[10:41:44] Begin /home/myuser/dev/pi-gen
[10:41:44] Begin /home/myuser/dev/pi-gen/stage0
[10:41:44] Unloading image
```

The devices do exist:
```
$ ls -l /dev/mapper/nbd10*
lrwxrwxrwx 1 root root 7 Out  1 10:41 /dev/mapper/nbd10p1 -> ../dm-4
lrwxrwxrwx 1 root root 7 Out  1 10:41 /dev/mapper/nbd10p2 -> ../dm-5

$ file /dev/dm-{4,5}
/dev/dm-4: block special (253/4)
/dev/dm-5: block special (253/5)

$ ls -l /dev/nbd10*
brw-rw---- 1 root disk 43, 320 Out  1 10:41 /dev/nbd10
brw-rw---- 1 root disk 43, 321 Out  1 10:41 /dev/nbd10p1
brw-rw---- 1 root disk 43, 322 Out  1 10:41 /dev/nbd10p2
```

Shouldn't `/dev/mapper/nbd10p1` be a partition of `/dev/mapper/nbd10` instead of a device?

Josh798, 2021-10-04:
Hi, I experienced this same issue. It seems that there is a delay between the time when the `kpartx` command at this line [here](https://github.com/RPi-Distro/pi-gen/blob/dcfd74d7d1fa293065ac6d565711e9ff891fe2b8/scripts/qcow2_handling#L111) finishes and the time at which the file in `$MAP_BOOT_DEV` (which is uses as soon as `kpartx` finishes) actually exists.

I used this patch on `scripts/qcow2_handling` and all appears to work:

```diff
diff --git a/scripts/qcow2_handling b/scripts/qcow2_handling
index f722937..b57cfaf 100644
--- a/scripts/qcow2_handling
+++ b/scripts/qcow2_handling
@@ -109,6 +109,11 @@ load_qimage() {
 EOF
                                sync
                                kpartx -a $NBD_DEV
+                               until [ -e $MAP_BOOT_DEV ]
+                               do
+                                       echo "Waiting for $MAP_BOOT_DEV to exist..."
+                               done
+                               echo "Found $MAP_BOOT_DEV."
                                mkdosfs -n boot -F 32 -v $MAP_BOOT_DEV
                                mkfs.ext4 -L rootfs -O "^huge_file,^metadata_csum,^64bit" $MAP_ROOT_DEV
                                sync
```

Obviously this is sort of a hack, but it works for me for now. I now get an error on the following line, but I think it's unrelated...

Josh798, 2021-10-05:
Actually, it looks like `kpartx` has an option to not return until all partitions are created. From `man kpartx`:

```
OPTIONS
       ...
       -s     Sync mode. Don't return until the partitions are created
```

So, I replaced all occurrences of `kpartx -a` in scripts/qow2_handling with `kpartx -as` and this seemed to fix the issue.

XECDesign, 2021-10-05:
Does Josh's fix resolve the issue, @1000283?