RPi-Distro/pi-gen #551: npm install with serialport fails

gcohler, 2021-10-01:
I'm trying to build a custom image.  When I go to npm install a program that uses serialport, the build fails.  I have tried adding the nan module in a variety of ways, but none works.  I believe that node-gyp is simply not working properly for some reason.

Here's what I'm trying.  Any advice welcome.

snippet of my script where the package.json uses serialport@9.2.4 -- which will require a node-gyp build because there is no pi image there.

```
on_chroot << EOF
sudo -g \#1001 -u \#1001 npm ci
```

and here's what happens:

```
> @serialport/bindings@9.2.4 install /home/<redacted>/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

qemu: Unsupported syscall: 397
prebuild-install WARN install No prebuilt binaries found (target=14.18.0 runtime=node arch=arm libc= platform=linux)
qemu: Unsupported syscall: 397
qemu: Unsupported syscall: 397
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:282:12)
gyp ERR! System Linux 4.15.0-159-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/<redacted>/node_modules/@serialport/bindings
gyp ERR! node -v v14.18.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.2.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@9.2.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/<redacted>/.npm/_logs/2021-10-01T10_47_24_536Z-debug.log
```

gcohler, 2021-10-01:
My theory is that this in fact has nothing to do with serial port -- but rather with running a node-gyp build in chroot.  That being said, this used to work with older stretch versions of pi-gen -- but it's been 18 mos or so since my last build.

XECDesign, 2021-10-03:
Sorry, can't help with anything npm-related. It could be an issue with qemu or just node not running well in a chroot, but I don't think there's anything particular about pi-gen itself which would cause issues like this.

gcohler, 2021-10-03:
I agree.  I was just hoping that you might have seen it.  I did figure out a work-around (that I don't love).  But such is life.

closed 2021-10-03 by gcohler