RPi-Distro/pi-gen #553: apt-get update not resolving in custom sub stage

MaDOS, 2021-10-14:
Hello,
I have set up pi gen on a WSL2 debian with a modified kernel to include nbd. 
I have also already managed to successfully generate a standard raspberry pi os lite image (Up to stage2) with this setup.
I was able to flash this on a CM4 and set up everything I need manually and everything seems to be working fine.

Now I wanted to add some custom stuff to my image. Mainly I want to add the azure iotedge runtime. 
To do this I want to do the following at the end of stage2:
- Add microsofts repository to my apt sources (for the moby and iotedge packages)
- Execute apt-get update
- Install moby and iotedge packages

I added a sub stage inside my stage2 named `04-edge-install`.
In there I have:
- a `files` folder with my `microsoft-prod.list` and the `microsoft.gpg.key`
- my run script named `00-run.sh`
- and a package listing named `01-packages`

**File contents:**

`microsoft.prod.list`:
```
deb [arch=amd64,arm64,armhf] http://packages.microsoft.com/debian/stretch/multiarch/prod stretch main
```

`microsoft.gpg.key`: (binary format) (When importing this on my debian with `apt-key` there was no problem with it being in binary fomat)

`00-run.sh`:
```
#!/bin/bash
install -m 644 files/microsoft-prod.list "${ROOTFS_DIR}/etc/apt/sources.list.d/"

on_chroot apt-key add - < files/microsoft.gpg.key
on_chroot << EOF
apt-get update
EOF
```

`01-packages`:
```
moby-engine moby-cli libssl1.0.2
iotedge
```

`config`
```
RELEASE=buster
DEPLOY_ZIP=0
ENABLE_SSH=1
STAGE_LIST="stage0 stage1 stage2"
```

**Probem**:
When running this with `sudo ./build-docker.sh -c config` I get to my substage 2-04 but suddenly apt refuses to resolve hostnames:

Log:
```
[16:10:13] End /pi-gen/stage2/03-set-timezone
[16:10:13] Begin /pi-gen/stage2/04-edge-install
[16:10:13] Begin /pi-gen/stage2/04-edge-install/00-run.sh
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
Err:1 http://packages.microsoft.com/debian/stretch/multiarch/prod stretch InRelease
  Temporary failure resolving 'packages.microsoft.com'
Err:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Temporary failure resolving 'raspbian.raspberrypi.org'
Hit:3 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
W: Failed to fetch http://packages.microsoft.com/debian/stretch/multiarch/prod/dists/stretch/InRelease  Temporary failure resolving 'packages.microsoft.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
[16:10:38] End /pi-gen/stage2/04-edge-install/00-run.sh
[16:10:38] Begin /pi-gen/stage2/04-edge-install/01-packages
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package moby-engine
E: Unable to locate package moby-cli
E: Unable to locate package iotedge
[16:10:39] Unloading image
```

This is at about 15-16minutes in and I have checked the log and can confirm that in stage0-00 `apt-get update` is working fine.

**Side-notes**:
I also had it fail sooner 3 times with the same resolving problem in stage2-02 when trying to get a package. (Which why i had suspected it might be a rate limiting problem or something but in 95% of cases it will fail at my custom stage)

It will also fail on my `apt-get update` when i change `on_chroot << EOF apt-get update EOF` to `on_chroot apt-get update` with the following:
```
[15:51:28] Begin /pi-gen/stage2/04-edge-install
[15:51:28] Begin /pi-gen/stage2/04-edge-install/00-run.sh
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
/usr/bin/apt-get: /usr/bin/apt-get: cannot execute binary file
[15:51:30] Unloading image
```

I am also currently running pi-gen on a freshly set up ubuntu 20.04 arm64 on a spare pi4b to see if it will fail in the same way (Will report this as soon when it will eventually finish). But this is marginally slower so I would really like to get it to work on a WSL setup.

MaDOS, 2021-10-15:
After a lot of digging and trial and error I figured out wat is going wrong here. This is related to recent problems with WSL2 overwriting /etc/resolv.conf.

I do not know why apt works in stage0 and then suddenly stops resolving hostnames correctly in stage2 but my fix was adding an entry to the static /etc/resolve.conf in the chroot environment.

So my final working setup looks like this:
- WSL2 debian system with a custom kernel that has nbd enabled (for anyone interested in doing this setup here is how to compile the kernel: https://www.frakkingsweet.com/running-pi-gen-on-wsl2/)
- Comment `modprobe nbd max_part=16` in qcow2_handling script so the kernel built-in nbd will be used
- stage2
| - 04-edge-install
  | - files
    | - microsoft-prod.list
    | - microsoft.gpg
  | - 00-run.sh
  | - 01-packages
  | - 02-packages
 
**microsoft-prod.list**:
```
deb [arch=amd64,arm64,armhf] https://packages.microsoft.com/debian/stretch/multiarch/prod stretch main
```

**00-run.sh**:
```
#!/bin/bash -e
install -m 644 files/microsoft-prod.list "${ROOTFS_DIR}/etc/apt/sources.list.d/"
on_chroot apt-key add - < files/microsoft.gpg

on_chroot << EOF
echo 'nameserver 1.1.1.1' | tee -a /etc/resolv.conf
apt-get -y update
EOF
```

**01-packages**:
```
moby-engine
```

**02-packages**:
```
aziot-edge
```

For me this issue is now solved. But if anyone has an idea why the resolve starts to fail somewhere between stage0 and stage2 it for sure would be intersting to know why.

bradisbell, 2023-03-09:
Were you ever able to find a more full explanation for this?  I'm having the same issue under Ubuntu under WSL2.

k-gordon, 2023-03-10:
> Were you ever able to find a more full explanation for this? I'm having the same issue under Ubuntu under WSL2.

He seems to have solved his issue, which was my own issue as well under WSL

> After a lot of digging and trial and error I figured out wat is going wrong here. This is related to recent problems with WSL2 overwriting /etc/resolv.conf.

Modifying `/etc/resolv.conf` with a different nameserver like `nameserver 1.1.1.1` did the trick for me.

bradisbell, 2023-03-28:
I think this has to do with IPv6 not supported under WSL2.  Still digging, but maybe that info helps someone else.