RPi-Distro/pi-gen #561: Module binfmt_misc not loaded in host

rumengb, 2021-11-09:
Hello I am using pi-gen to build a custom RPI distribution dedicated to astronomy, but since the switch to bullseye it does not work for me anymore. I tried to use this commit  #a8596e0d894315396d62d3546ce96cd1e7ceaf75 (the last one before the switch) and everything worked fine again but with the next one I started to get this error:

Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc

I an using Ubuntu 20.04 I also tried Ubuntu 21.04 same results.

cookew, 2021-11-09:
Are you running this in Docker with the build-docker.sh script? I ran into this issue while running it in Docker. After manually running the Docker build image, I could in fact see that the binfmt_misc module is loaded by running `lsmod | binfmt_misc` in the container, but the /proc/sys/fs/binfmt_misc mount is not mounted, which the dependencies_check script is actually checking for. The /proc/sys/fs/binfmt_misc is mounted on the hose (outside of the Docker container). To fix it, I ran the following command to replace the `exit 1` line in the dependencies_check script to mount what it needs.

`sed -i 's@exit 1@mount -t binfmt_misc binfmt_misc /proc/sys/fs/binfmt_misc@g' scripts/dependencies_check`

rumengb, 2021-11-09:
This is what happened after the command:

WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc
[13:31:58] Begin /pi-gen
[13:31:58] Begin /pi-gen/stage0
Creating base image: image-stage0.qcow2
Formatting 'image-stage0.qcow2', fmt=qcow2 cluster_size=65536 extended_l2=off compression_type=zlib size=12884901888 lazy_refcounts=off refcount_bits=16
/pi-gen/scripts/qcow2_handling: line 106: sfdisk: command not found
[13:31:58] Unloading image

real	0m1,605s
user	0m0,031s
sys	0m0,037s

rumengb, 2021-11-09:
there is a little progress:

Successfully built 8e7c5c3ed814
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc
[14:44:02] Begin /pi-gen
[14:44:02] Begin /pi-gen/stage0
[14:44:02] Begin /pi-gen/stage0/prerun.sh
W: qemu-debootstrap is deprecated. Please use regular debootstrap directly
I: Running command: debootstrap --arch armhf --components main,contrib,non-free --keyring /pi-gen/stage0/files/raspberrypi.gpg --exclude=info bullseye /pi-gen/work/indigosky/stage0/rootfs http://raspbian.raspberrypi.org/raspbian/
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id A0DA38D0D76E8B5D638872819165938D90FDDD2E)
I: Retrieving Packages 
I: Validating Packages 
I: Retrieving Packages 
........
I: Configuring gnupg...
I: Configuring raspbian-archive-keyring...
I: Configuring apt...
I: Configuring apt-utils...
I: Configuring tasksel...
I: Configuring tasksel-data...
I: Configuring libc-bin...
I: Base system installed successfully.

I m getting to this point and it stays here for hours... and never gets beyond...I have to restart docker inorder to stop and remove the container...

jketterl, 2021-11-11:
I've run into the same problem, on docker. I used to test changes on my dev machine usint the docker method before pushing them. The automated build system uses ARM EC2 instances (without docker) and is still working fine.

I haven't really been able to find a conclusion yet, but here's what I found:

- The error message is ambiguous in the docker context because it is referring to the "host", which could be both the docker host or the docker container. The check itself is done inside the docker container, but the failure may be on both the docker host or container level.
- `binfmt_misc` is a kernel feature that allows the registration of userspace applications to run specific binary formats. In the case of pi-gen, it is used to be able to run ARM binaries on an x86 host with the use of `qemu`.
- Interesting discussion of a similiar bug: https://github.com/containers/podman/issues/4725
- Using `binfmt_misc` inside a (docker) container seems to be problematic in itself. This kernel subsystem does not (yet?) support namespaces, and as such any changes made on either the host or container level are global and persistent. If a container registers a userspace application, that registration will be effective for the host or any other container on that host. The registration will survive the container that registered it (I haven't checked if it would still viable though given that the filesystem the registered binary resides in does not).
- On Debian, the registration of `qemu` is handled by the `qemu-user-static` package, more specifically its postinst script.
- pi-gen has extra calls to that [here](https://github.com/RPi-Distro/pi-gen/blob/master/build-docker.sh#L100) and [here](https://github.com/RPi-Distro/pi-gen/blob/master/build-docker.sh#L114), making sure that the registration is in place before running the actual build (and potentially updating it? I haven't checked if the command changes things if it encounters an existing registration).
- The postinst script of the Buster version of that package tried to mount the `binfmt_misc` filesystem (that will be checked by the `dependencies_check` script), and it would succeed in the case of pi-gen since the container is run with `--privileged`.
- The postinst script of the Bullseye version does not seem to provide that mountpoint any longer, however still manages to register `qemu` in some way. This is probably the reason why this issue is now showing up.
- Both versions of the postinst script seem to try to detect if they are run inside a container, and not perform any actions in that case (probably because of the namespacing problem mentioned above). The checks themselves however seem to be broken / not aimed at docker and as such do not prevent the scripts from performing the registration. It's probably a bad idea to rely on that.

With all that information, I was still unable to get my local builds to work again. I've tried both an i386 container and the approach to use i386 qemu inside an amd64 container. Both of them get to a certain point and fail with this:

```
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  At least one invalid signature was encountered.
Err:2 http://archive.raspberrypi.org/debian bullseye InRelease
  At least one invalid signature was encountered.
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://raspbian.raspberrypi.org/raspbian bullseye InRelease: At least one invalid signature was encountered.
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
```

Immediately reminded me of #271 except that this is inside the virtualization now? I'm not sure if this is still related to `binfmt_misc` any more or if qemu in Bullseye has a new bug now...

buanet, 2021-11-12:
Hey guys, same story here. :) Looks like it is actually not working because of binfmt_misc is not loaded automatically in debian:bullseye based containers. So the [depndencies_check](https://github.com/RPi-Distro/pi-gen/blob/6117b9bb6193e377d6bf99d5d94c889525bc1a34/scripts/dependencies_check#L45) fails and gives this message: 

```
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc
```

 Did some googling about that, but no solution so far... 

@jketterl I'm pretty sure your GPG error comes from an old libseccomp2 version. Check out [this issue](https://github.com/buanet/ioBroker.docker/issues/199#issuecomment-9329187809) and make sure you got libseccomp2 version > v2.4 installed on your host. As far as I know this issue is alro related to debian bullseye in combination with arm/ i386 architecture...

Regards,
André

jketterl, 2021-11-12:
The `libseccomp` version on my host is 2.5.2, so should be good. The host is running Manjaro, not Debian, so it's usually up to date.

buanet, 2021-11-12:
@jketterl kk, I ran into this issue this morning before the binfmt_misc thing... so I thought I would help you too... :)

rumengb, 2021-11-12:
Hm... I an having this problem and i and with:
Package: libseccomp2
Version: 2.5.1-1ubuntu1~20.04.1

buanet, 2021-11-14:
@rumengb The libseccomp2 thing in only interesting for those who get an  GPG error like jketterl wrote above. I got the same error and solved it with updating the package on the build host. 

Your original issue `Module binfmt_misc not loaded in host` is, as you already noticed, based on the bullseye change. I did some more testing and got it working again when switching the base image of the Docker container which is used to run the pi-gen build.sh back to buster.
I did this by simply replacing `:bullseye` with `:buster` in `build-docker.sh` and `Dockerfile` before running `build-docker.sh`.
```
sed -i 's/:bullseye/:buster/' ./build-docker.sh
sed -i 's/:bullseye/:buster/' ./Dockerfile
```
This brought my ci script back to life an generated a bootable raspberry os image. 

So for me it definitely looks like there is a general problem with `qemu-user-static` and `binfmt_misc` in bullseye. 
Today I spend some hours searching and reading about that topic to find a way how to fix this but with no success so far. :( 

Regards,
André

jsmouret, 2021-11-14:
I had it working with:
. upgrade host to bullseye => fix some ldconfig crash
. use arm64v8/debian:bullseye image => bypass the binfmt check

jketterl, 2021-11-15:
> @rumengb The libseccomp2 thing in only interesting for those who get an GPG error like jketterl wrote above. I got the same error and solved it with updating the package on the build host.

As I said, I already am on version 2.5.2 of libseccomp, and after reading the release notes for 2.5.3 (latest release) I don't think it would fix my problem.

> So for me it definitely looks like there is a general problem with `qemu-user-static` and `binfmt_misc` in bullseye.

For me this looks more like the filesystem mount check is no longer applicable. The registration of the binary formats seems to be in place (the postinst script still works, it just doesn't provide the mountpoint any more), so the execution of ARM binaries should work. I don't see that mount point used other than for the check. Not sure what other check would work though.

jketterl, 2021-11-15:
Also, one more thing I just noticed... The `build-docker.sh` runs the actual build in a container with the `--privileged` flag, which disables seccomp. I think seccomp is a red herring here.

jketterl, 2021-11-16:
Looks like I was wrong about the mountpoint no longer being necessary. I had the mountpoint patched into the `build-docker.sh` script at some point to trick the dependency check, but once I remove it, the registration seems to be skipped as well. I've been fooled by the global state, it's easy to think everything works... until you reboot. So maybe the correct conclusion would be to have `build-docker.sh` set up the mountpoint on container start.

rubencabrera, 2021-12-17:
In need for a quick fix for my pipeline, I followed @buanet's [suggestion](https://github.com/RPi-Distro/pi-gen/issues/561#issuecomment-968180324), replaced bullseye with buster and got a working image after removing the `python-is-python3` package from stage2. 

Sorry for bumping this thread with something that's not a solution to the root cause, but there are a few open issues around this problem and none provided a workaround that worked for me.

XECDesign, 2022-01-31:
Should be resolved with https://github.com/RPi-Distro/pi-gen/pull/581 ?