RPi-Distro/pi-gen #562: Ubuntu 16.04.8 libarchive13

myxxmikeyxx, 2021-11-09:
I am getting `The following packages have unmet dependencies:
 libarchive-tools : Depends: libarchive13 (= 3.2.1-2~ubuntu16.04.1) but 3.1.2-11ubuntu0.16.04.8 is to be installed
E: Unable to correct problems, you have held broken packages.`
When I try to install libarchive-tools. Anyone know a work around? 
I don't want to download the .deb file and modify it to allow for the version I'm on but I don't know another way.
Reference for modifying .deb files [here](https://unix.stackexchange.com/questions/138188/easily-unpack-deb-edit-postinst-and-repack-deb)

erikkugel, 2021-11-24:
It looks like you can just get bsdtar from the bsdtar package, that appears to be the dependency that libarchive provides:

`sudo apt-get install bsdtar`

alexander-sony, 2022-12-06:
> It looks like you can just get bsdtar from the bsdtar package, that appears to be the dependency that libarchive provides:
> 
> `sudo apt-get install bsdtar`

Facing the same problem. Do you mean that `bsdtar` can be used instead of `libarchive-tools`?

XECDesign, 2022-12-12:
Yes, bsdtar used to have its own separate package, but is now a part of libarchive-tools. So for an old OS, you should use bsdtar. However, I don't think Ubuntu 16.04 is likely to work in the first place, unless you pull in a much newer version of qemu than what's available in the repo.

myxxmikeyxx, 2022-12-12:
I have since moved on from this project and found a different device to meet the needs I had. 
I am sad to not be using raspberry pi's at work (even after we got two kits), but when you work for a place willing to throw money at problems and not customize solutions, that's how it goes. 

@XECDesign What OS do you recommend? The only reason (at the time) I was using Ubuntu 16 because I was also trying buildroot which at the time recommend ubuntu 16 still.

XECDesign, 2022-12-12:
Debian Bullseye seems to be okay. I haven't tested recent Ubuntu releases properly yet.

myxxmikeyxx, 2022-12-12:
No problem. I will close this out with that info.
Thank you everyone for the input.

closed 2022-12-12 by myxxmikeyxx