RPi-Distro/pi-gen #567: binfmt_misc mount exists on the host system but not from inside the docker container

MorganBorman, 2021-11-23:
> Is the issue that the mount exists on the host system but not from inside the docker container? If that's the case, I'm not sure what the right solution is.

_Originally posted by @XECDesign in https://github.com/RPi-Distro/pi-gen/issues/264#issuecomment-514210365_

MorganBorman, 2021-11-23:
This is the issue I'm seeing.

On my host system;

```sh
$ uname -a
Linux myhostname 5.15.4-arch1-1 #1 SMP PREEMPT Sun, 21 Nov 2021 21:34:33 +0000 x86_64 GNU/Linux
$ lsb_release -a
LSB Version:    1.4
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a
$ grep "/proc/sys/fs/binfmt_misc" /proc/mounts
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=11867 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,nosuid,nodev,noexec,relatime 0 0
```

I modified `scripts/dependencies_check` to print out the mounts;

```diff
$ git diff
diff --git a/scripts/dependencies_check b/scripts/dependencies_check
index 32c42ef..0d429fe 100644
--- a/scripts/dependencies_check
+++ b/scripts/dependencies_check
@@ -43,6 +43,7 @@ dependencies_check()
        esac
 
        if [[ "${binfmt_misc_required}" == "1" ]]; then
+               cat /proc/mounts
                if ! grep -q "/proc/sys/fs/binfmt_misc" /proc/mounts; then
                        echo "Module binfmt_misc not loaded in host"
                        echo "Please run:"
```

Within the container that outputs;

```
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
cgroup /sys/fs/cgroup cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot 0 0
/dev/mapper/fasthome /config ext4 ro,relatime,data=ordered 0 0
dev /dev devtmpfs rw,nosuid,relatime,size=32886952k,nr_inodes=8221738,mode=755,inode64 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev,inode64 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
hugetlbfs /dev/hugepages hugetlbfs rw,relatime,pagesize=2M 0 0
mqueue /dev/mqueue mqueue rw,nosuid,nodev,noexec,relatime 0 0
/dev/nvme1n1p2 /pi-gen/deploy ext4 rw,relatime 0 0
/dev/nvme1n1p2 /pi-gen/work ext4 rw,relatime 0 0
/dev/nvme1n1p2 /lib/modules ext4 rw,relatime 0 0
/dev/nvme1n1p2 /etc/resolv.conf ext4 rw,relatime 0 0
/dev/nvme1n1p2 /etc/hostname ext4 rw,relatime 0 0
/dev/nvme1n1p2 /etc/hosts ext4 rw,relatime 0 0
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc
```

jketterl, 2021-11-24:
See also #561 - the title does not mention it, but that issue is about docker as well, looks like a duplicate to me.

MorganBorman, 2021-11-30:
@jketterl Agreed there's probably some duplication here. I actually read your issue before opening this one - specifically, your [summary comment](https://github.com/RPi-Distro/pi-gen/issues/561#issuecomment-966336861) was super helpful!

I wanted to open this issue (even if it gets closed as a duplicate) to document what I am seeing and the work-around(s) I'm using - since #561 has gotten a bit into the weeds.

I've been working off 288b21fc27e128ea6b330777aca68e0061ebf4fe

I made the following changes (*includes the change from #558*);

```patch
diff --git a/Dockerfile b/Dockerfile
index e7fc561..479e708 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -8,7 +8,7 @@ RUN apt-get -y update && \
         git vim parted \
         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools \
         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc\
-        binfmt-support ca-certificates qemu-utils kpartx \
+        binfmt-support ca-certificates qemu-utils kpartx fdisk \
     && rm -rf /var/lib/apt/lists/*
 
 COPY . /pi-gen/
diff --git a/scripts/dependencies_check b/scripts/dependencies_check
index 32c42ef..4b698c8 100644
--- a/scripts/dependencies_check
+++ b/scripts/dependencies_check
@@ -43,7 +43,8 @@ dependencies_check()
        esac
 
        if [[ "${binfmt_misc_required}" == "1" ]]; then
-               if ! grep -q "/proc/sys/fs/binfmt_misc" /proc/mounts; then
+               mount binfmt_misc -t binfmt_misc /proc/sys/fs/binfmt_misc
+               if (! grep -q "/proc/sys/fs/binfmt_misc" /proc/mounts) && (! lsmod | grep -q binfmt_misc); then
                        echo "Module binfmt_misc not loaded in host"
                        echo "Please run:"
                        echo "  sudo modprobe binfmt_misc"
diff --git a/scripts/qcow2_handling b/scripts/qcow2_handling
index 6b2a56b..539ca54 100644
--- a/scripts/qcow2_handling
+++ b/scripts/qcow2_handling
@@ -16,7 +16,9 @@ export MAP_ROOT_DEV
 
 # find and initialize free block device nodes
 init_nbd() {
-       modprobe nbd max_part=16
+       if [ ! -d "/sys/class/block/nbd0" ]; then
+               modprobe -v nbd max_part=16
+       fi
        if [ -z "${NBD_DEV}" ]; then
        for x in /sys/class/block/nbd* ; do
                S=`cat $x/size`
```

That doesn't completely fix the build because I think `modprobe -v nbd max_part=16` tries to load the arm version of the `nbd` kernel module;

```
insmod /lib/modules/5.15.4-arch1-1/kernel/drivers/block/nbd.ko.zst max_part=16
modprobe: ERROR: could not insert 'nbd': Exec format error
```

I work around that by adding the `nbd` module from my host system first;

```sh
sudo modprobe -v nbd max_part=16
```

After which the `build-docker.sh` build script succeeds.

pinkforest, 2021-12-02:
@EdwardCooke I hit this same issue and your work/around fixed build-docker.sh for me as well. Ubuntu 20.04.2 LTS

jketterl, 2021-12-06:
I have found a temporary solution on my end which involves installing qemu and it's binfmt_misc registration on the host, and removing the corresponding parts from the container setup. I wouldn't consider this a solution since this makes the build dependent on the host environment (which is pretty much what I wanted to avoid when I went for the docker build) but given that this would pollute my host anyway it doesn't really matter anyway. Also, I do understand that there's no way to properly encapsulate anything involving `binfmt_misc` in a container since the kernel just doesn't support it.

Here's my patches for anyone who would like to try the same:

```diff
diff --git a/build-docker.sh b/build-docker.sh
index a7c23e3..d1f95fe 100755
--- a/build-docker.sh
+++ b/build-docker.sh
@@ -96,8 +96,7 @@ if [ "${CONTAINER_EXISTS}" != "" ]; then
                --env-file "${CONFIG_FILE}" \
                -e "GIT_HASH=${GIT_HASH}" \
                pi-gen \
-               bash -e -o pipefail -c "dpkg-reconfigure qemu-user-static &&
-       cd /pi-gen; ./build.sh ${BUILD_OPTS} &&
+               bash -e -o pipefail -c "cd /pi-gen; ./build.sh ${BUILD_OPTS} &&
        rsync -av work/*/build.log deploy/" &
        wait "$!"
 else
@@ -110,8 +109,7 @@ else
                --env-file "${CONFIG_FILE}" \
                -e "GIT_HASH=${GIT_HASH}" \
                pi-gen \
-               bash -e -o pipefail -c "dpkg-reconfigure qemu-user-static &&
-       cd /pi-gen; ./build.sh ${BUILD_OPTS} &&
+               bash -e -o pipefail -c "cd /pi-gen; ./build.sh ${BUILD_OPTS} &&
        rsync -av work/*/build.log deploy/" &
        wait "$!"
 fi
diff --git a/scripts/dependencies_check b/scripts/dependencies_check
index 32c42ef..eb35f05 100644
--- a/scripts/dependencies_check
+++ b/scripts/dependencies_check
@@ -28,26 +28,4 @@ dependencies_check()
                false
        fi
 
-    # If we're building on a native arm platform, we don't need to check for
-    # binfmt_misc or require it to be loaded.
-
-       binfmt_misc_required=1
-
-       case $(uname -m) in
-               aarch64)
-                       binfmt_misc_required=0
-                       ;;
-               arm*)
-                       binfmt_misc_required=0
-                       ;;
-       esac
-
-       if [[ "${binfmt_misc_required}" == "1" ]]; then
-               if ! grep -q "/proc/sys/fs/binfmt_misc" /proc/mounts; then
-                       echo "Module binfmt_misc not loaded in host"
-                       echo "Please run:"
-                       echo "  sudo modprobe binfmt_misc"
-                       exit 1
-               fi
-       fi
 }
```

The procedure to setup qemu on the host is distribution-specific; I think for Debian it's done by simply installing the `qemu-user-static` packages, but I haven't tested that.

zoff99, 2022-07-17:
i am having sort of the same issue when running build_docker.sh

```chroot: failed to run command '/bin/true': Exec format error```

any updates on this?

pinkforest, 2022-07-17:
That's a different issue me thinks... I suspect something else now might broken and I think needs a new issue

This issue should have been fixed here by @XECDesign merged into master 1 Feb

https://github.com/RPi-Distro/pi-gen/pull/581

And further mentioned here:

https://github.com/RPi-Distro/pi-gen/issues/561#issuecomment-1025927900

Did you pull pi-gen from the git master branch?

pinkforest, 2022-07-17:
@MorganBorman and @jketterl can we confirm fix by #581 and close if fixed - I have to find some time to play with this again

jketterl, 2022-07-17:
Well, what can I say. I can confirm that the error message from the dependencies check does no longer appear for me. It's not really much help though since I am still unable to build images using the docker method.

I had "fixed" the problem for now by installing the corresponding `qemu-user-static` registration on my (Manjaro) base system, so in order to perform this test, I removed this package again and rebooted my system. Result:

```
[...]
I: Extracting ncurses-base...
I: Extracting ncurses-bin...
I: Extracting passwd...
I: Extracting perl-base...
I: Extracting sed...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot "/pi-gen/work/test/stage0/rootfs" /bin/true
W: See /pi-gen/work/test/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/test/stage0/rootfs/debootstrap': Directory not empty
[14:31:12] bootstrap failed: please check /pi-gen/work/test/stage0/debootstrap.log
```

last lines of `debootstrap.log`

```
2022-07-17 14:31:06 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/t/tar/tar_1.34+dfsg-1_armhf.deb [798860/798860] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/tar_1.34+dfsg-1_armhf.deb" [1]
2022-07-17 14:31:06 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/t/tasksel/tasksel_3.68+deb11u1_all.deb [100928/100928] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/tasksel_3.68+deb11u1_all.deb" [1]
2022-07-17 14:31:07 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/t/tasksel/tasksel-data_3.68+deb11u1_all.deb [18016/18016] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/tasksel-data_3.68+deb11u1_all.deb" [1]
2022-07-17 14:31:07 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/t/traceroute/traceroute_2.1.0-2_armhf.deb [51310/51310] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/traceroute_1%3a2.1.0-2_armhf.deb" [1]
2022-07-17 14:31:08 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/t/tzdata/tzdata_2021a-1+deb11u4_all.deb [284760/284760] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/tzdata_2021a-1+deb11u4_all.deb" [1]
2022-07-17 14:31:08 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/s/systemd/udev_247.3-7+rpi1_armhf.deb [1430256/1430256] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/udev_247.3-7+rpi1_armhf.deb" [1]
2022-07-17 14:31:09 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/u/util-linux/util-linux_2.36.1-8+deb11u1_armhf.deb [1037516/1037516] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/util-linux_2.36.1-8+deb11u1_armhf.deb" [1]
2022-07-17 14:31:09 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/v/vim/vim-common_8.2.2434-3+deb11u1_all.deb [226176/226176] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/vim-common_2%3a8.2.2434-3+deb11u1_all.deb" [1]
2022-07-17 14:31:10 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/v/vim/vim-tiny_8.2.2434-3+deb11u1_armhf.deb [600948/600948] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/vim-tiny_2%3a8.2.2434-3+deb11u1_armhf.deb" [1]
2022-07-17 14:31:10 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/v/vim/xxd_8.2.2434-3+deb11u1_armhf.deb [191608/191608] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/xxd_2%3a8.2.2434-3+deb11u1_armhf.deb" [1]
2022-07-17 14:31:11 URL:http://debian.bio.lmu.de/raspbian/raspbian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2+deb11u1_armhf.deb [84700/84700] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2+deb11u1_armhf.deb" [1]
chroot: failed to run command '/bin/true': Exec format error
```

Looking at the binfmt registrations, this is not surprising: There was no regsitration for any ARM binaries on my system at this point, so the system simply was not properly prepared to run this binary.

I haven't done any further analysis since all of these observations are in line with my previous analysis, see: https://github.com/RPi-Distro/pi-gen/issues/561#issuecomment-966336861