RPi-Distro/pi-gen #568: Shared environment variables between stages

chuongle, 2021-11-24:
It looks like only default environment variables can be used across stages. Is there a solution to define a variable in the `config` or `build.sh` and reuse it across stages?

XECDesign, 2021-11-24:
Not that I'm aware of, unfortunately. I had to do something similar using this somewhat ugly way:

https://github.com/RPi-Distro/pi-gen/blob/2f12cf985a88977957e2743601246d2a07aeb660/export-noobs/prerun.sh#L61

https://github.com/RPi-Distro/pi-gen/blob/2f12cf985a88977957e2743601246d2a07aeb660/export-noobs/00-release/00-run.sh#L40

chuongle, 2021-11-24:
Hi @XECDesign! Thank you for getting back about this. I think I can try this.

However I only want to build the minimal image so I only have `stage 0 stage 1 stage 2` in my `STAGE_LIST`. Do you think I can do this?

- stage 0: i will create a `prerun.sh` something like this
```
echo "CUSTOM_ENV" > "/etc/variables" 
```
- stage 1/2: i will pull that variable out to use in these stages?

I appreciate your help on this. Thanks!

XECDesign, 2021-11-24:
Yeah, but I think you'll need to pull the variable out in each file script you want to use it, which isn't pretty.

Does the variable change between builds? If not, you could put `export SOMETHING=foo` in `config` ?

chuongle, 2021-11-24:
The variable stay the same through out stages but only different when building as Production vs Development image. I did not know `export SOMETHING=foo` is available in `config`. I will try that out. Thanks!

chuongle, 2022-01-08:
I figured out how to do this. Posting here in case someone else encounter the same issue.

So for my case, I need to pull different assets depending on the environment (staging/production). So far I only need this environment in 3 tasks for stage2. So I end up adding this in my `build.sh`

``

closed 2022-01-08 by chuongle