RPi-Distro/pi-gen #569: failed to extract zliblg? chroot: failed to run command ‘/debootstrap/debootstrap’:

Sheng-IoT, 2021-12-06:
Hi, I switched to the amr64 branch and then ran a clean build, but I got the following error, do you think the issue is caused by failing to extract zliblg (maybe related to an unstable internet environment)? I cloned the latest repo and works on a Debian 10 host machine. 

here is the error message:

>......
I: Extracting util-linux...
I: Extracting libxxhash0...
I: Extracting liblzma5...
I: Extracting zlib1g...
I: Running command: chroot /home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error
rmdir: failed to remove '/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs/debootstrap': Directory not empty
[10:48:14] bootstrap failed: please check /home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/debootstrap.log

and I checked the **pi-gen/work/<image name>/stage0/debootstrap.log** for detailed information, and I got this:

> .....
2021-12-06 10:48:07 URL:http://deb.debian.org/debian/pool/main/v/vim/vim-tiny_8.2.2434-3_arm64.deb [692844/692844] -> "/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs//var/cache/apt/archives/partial/vim-tiny_2%3a8.2.2434-3_arm64.deb" [1]
2021-12-06 10:48:08 URL:http://deb.debian.org/debian/pool/main/v/vim/xxd_8.2.2434-3_arm64.deb [191516/191516] -> "/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs//var/cache/apt/archives/partial/xxd_2%3a8.2.2434-3_arm64.deb" [1]
2021-12-06 10:48:08 URL:http://deb.debian.org/debian/pool/main/x/xxhash/libxxhash0_0.8.0-2_arm64.deb [18544/18544] -> "/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs//var/cache/apt/archives/partial/libxxhash0_0.8.0-2_arm64.deb" [1]
2021-12-06 10:48:08 URL:http://deb.debian.org/debian/pool/main/x/xz-utils/liblzma5_5.2.5-2_arm64.deb [164528/164528] -> "/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs//var/cache/apt/archives/partial/liblzma5_5.2.5-2_arm64.deb" [1]
2021-12-06 10:48:09 URL:http://deb.debian.org/debian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2_arm64.deb [87944/87944] -> "/home/sheng/workspace/pi-gen/work/pigen-on-debian-2.0/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2_arm64.deb" [1]

XECDesign, 2021-12-06:
Couple of things to check:

- [ ] `qemu-arm-static` is installed
- [ ] `/usr/bin/qemu-aarch64-static` is present
- [ ] `/proc/sys/fs/binfmt_misc/qemu-aarch64` exists and has something like this in it:
```
enabled
interpreter /usr/bin/qemu-aarch64-static
flags: OCF
offset 0
magic 7f454c460201010000000000000000000200b700
mask ffffffffffffff00fffffffffffffffffeffffff
```

Sheng-IoT, 2021-12-08:
Thank you for your response, I went through the checklist you mentioned, but I didn't see any problem. After that, I decided to switch to a Debian 11 host machine and start the build from scratch, and it works. I still don't know why it won't work on Debian 10, but I will close this issue for now as I find a "workaround". Thank you.

closed 2021-12-08 by Sheng-IoT