RPi-Distro/pi-gen #570: Target is busy

cdupont, 2021-12-07:
I often have this issue, toward the end of the compilation:
```
Disconnect block device
/dev/nbd0 disconnected
Shrinking qcow2 image
Image resized.
Convert qcow2 to raw image

Get PARTUUIDs from image
Boot: 7d4f58c4-01
Root: 7d4f58c4-02
Mount image
Setup PARTUUIDs
Set UUIDs to make it bootable
Umount image
umount: /var/lib/jenkins/workspace/5_ISO/work/WaziGate/tmpimage/boot: target is busy.
```

The script cannot umount the temporary image.
However, I can do it manually with `sudo umount /var/lib/jenkins/workspace/5_ISO/work/WaziGate/tmpimage/boot`.
Any idea?

XECDesign, 2021-12-08:
Adding qcow2 image support seems to have caused too many issues, so since https://github.com/RPi-Distro/pi-gen/commit/68fee340f95eb630a706e2dbf75615c6be4e980b it is marked as experimental and is definitely not recommended.

cdupont, 2021-12-08:
So not using qcow2 will be more reliable? What will be the performance hit?

XECDesign, 2021-12-08:
It will use a LOT more disk space.