RPi-Distro/pi-gen #574: What exactly is arm boost?

arnav7633, 2021-12-25:
I saw an option in the config.txt called the `arm_boost`. What is it exactly? Does it affect temps on idle? Does it mean that my pi would be continuously running on max clock speeds?

XECDesign, 2021-12-25:
https://www.raspberrypi.com/news/bullseye-bonus-1-8ghz-raspberry-pi-4/

https://www.raspberrypi.com/documentation/computers/config_txt.html#arm_boost-raspberry-pi-4-only

arnav7633, 2021-12-25:
What i wanted to know that my pi is idling at 52C in a case with a heatsink in a 20C ambient temp

XECDesign, 2021-12-25:
The setting shouldn't affect idle temperatures.

A heatsink relies on airflow to take the heat away, so having one inside a case is a good way to trap heat. Unless the case routes the heat out somehow or has a fan with a sensible airflow channel, this isn't an ideal setup.

For general support, please use https://www.raspberrypi.org/forums

closed 2021-12-25 by XECDesign