RPi-Distro/pi-gen #578: Error when compiling arm64 branch

fuji246, 2022-01-02:
```
I: Running command: chroot /home/ubuntu/pi-gen/work/lomorage/stage0/rootfs /debootstrap/debootstrap --second-stage
W: Failure trying to run:  /sbin/ldconfig
W: See //debootstrap/debootstrap.log for details
rmdir: failed to remove '/home/ubuntu/pi-gen/work/lomorage/stage0/rootfs/debootstrap': Directory not empty
[20:51:03] bootstrap failed: please check /home/ubuntu/pi-gen/work/lomorage/stage0/debootstrap.log
```
And debootstrap.log shows:

```
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
```

Env: Ubuntu 16.04 vm on Windows

clarkchentw, 2022-01-02:
I think I used to have this issue, then I upgraded from buster to bullseye. You can try to see if 20.04 will fix it.

XECDesign, 2022-01-04:
Sounds like something that might be caused by an old version of qemu, so trying a newer distro release is likely to be the solution.

xavieryin, 2023-06-01:
I faced similar issue starting in Windows 11 WSL2 Ubuntu 18.04, and somehow made it work eventually, here's what I did.

0. in WSL2 Ubuntu 18.04
  0.1. main / arm64 branch; build.sh / build-docker.sh => all failed, as several issues opened here
  0.2. `sudo apt-get update && sudo apt-get upgrade` and then repeat 0.1. => still failed
1. install WSL distros: Ubuntu 20.04, and then Debian by `wsl --install <distro name>`
2. in each of these two WSL distro
  2.1. `sudo apt-get update && sudo apt-get install <dependencies in README>`
  2.2. `git clone --depth 1 --branch arm64`
  2.3. `nano config,` and then `sudo build.sh` => both success !!!
3. now go back to WSL2 Ubuntu 18.04, **MAGICALLY**...
  3.1. `sudo rm -rf work`
  3.2. **`sudo apt-get update && sudo apt-get upgrade`**
  3.3. `sudo ./build.sh` => **NOW SUCCESS** (contrary to step 0)
4. repeated all above on another PC, same

I cannot explain this, but this is what actually happened. Folks working on Windows 11 may worth a try (YMMV). Perhaps these steps shed some light to the root cause of all these related issues.