RPi-Distro/pi-gen #580: 64 bit image

ThirtySomething, 2022-01-17:
Hi there,

is it somehow possible to build an image of a 64 bit Raspi OS? A 64 bit version of Raspi OS is available [here](https://downloads.raspberrypi.org/raspios_arm64/). During my tests I was not able to find some kind of switch to select between the usual 32 bit version and the new 64 bit version.

Regards

ThirtySomething

XECDesign, 2022-01-17:
Yup, it's built from the arm64 branch. Merging it so that everything is built from one branch is on the todo list.

ThirtySomething, 2022-01-17:
@XECDesign That's great, so I'm still waiting.

XECDesign, 2022-01-17:
Not sure what there's to wait for. The branch is and has been available to use for a long time.

ThirtySomething, 2022-01-19:
@XECDesign I'm sorry - there was a misunderstanding from my side. I've expected this in the same script, not in a different branch. Now it's clear from my side. Thanks for pushing me in the right direction.

closed 2022-01-19 by ThirtySomething

maxpavlov, 2023-02-28:
Hey. Just a quick on for 2023 - if one wants to build a 64bit image, the arm64 branch should be used still, right? If an image is created based on the sources from the master branch, it will be for 32bit OS, correct? Thanks :)

XECDesign, 2023-02-28:
That's right