RPi-Distro/pi-gen #586: Where is the kernel package's .config file hosted?

jonesmz, 2022-02-05:
I see that this builder grabs a binary package of the raspberrypi-kernel from the raspberrypi.org apt repo, but i can't figure out where that package's .config is hosted for the build.

Can you direct me to the right place?

My reason for asking is that I work on a Gentoo based image for the raspberrypi, and want to use the same config as this project for our kernel builds, but don't want to use the binary package. We want to build from source.

https://github.com/GenPi64

popcornmix, 2022-02-06:
[pi0/1](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/configs/bcmrpi_defconfig)
[pi2/3](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/configs/bcm2709_defconfig)
[pi4](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/configs/bcm2711_defconfig)
[64bit](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm64/configs/bcm2711_defconfig)

jonesmz, 2022-02-06:
Thank you.

The kernel packages are compiled with only that config?

Where is the build script for the kernel .deb so I can make sure I'm not skipping any important steps?

XECDesign, 2022-02-07:
The deb just ships pre-built files from https://github.com/raspberrypi/firmware/tree/stable and is not a good example of how to do anything.

Kernel packages tend to integrate quite closely with the distro to make sure headers work, various hooks run, initramfs is updated properly and so on.

It has been many years since I've run gentoo as my main OS, so I can't be of much help. Maybe a fork of gentoo-sources to pull from https://github.com/raspberrypi/linux/ and then something like genkernel might be a better starting point. Just don't forget to install the dtb and dtbo files.

popcornmix, 2022-02-07:
I'd expect a kernel built for arch/arm64/configs/bcm2711_defconfig to boot on Pi0w2 - it has the same processor as pi3.
Make sure your kernel tree and firmware are new enough and you are installing the dtb/dtbo files.

XECDesign, 2022-02-09:
Closing since it's not a pi-gen issue.

Do you have a link to the file that defines how to build the kernel and which files to install?

closed 2022-02-09 by XECDesign

jonesmz, 2022-02-13:
The kernel build is driven by this script:

https://github.com/GenPi64/genpi64-overlay/blob/master/sys-kernel/raspberrypi-kernel/raspberrypi-kernel-5.10.92.ebuild

Which references this script

https://github.com/GenPi64/genpi64-overlay/blob/master/eclass/pikernel-build.eclass

Which further references this script

https://github.com/gentoo/gentoo/blob/master/eclass/kernel-build.eclass