RPi-Distro/pi-gen #590: chroot: failed to run command '/bin/true': Exec format error

jnnks, 2022-02-16:
I am not able to build an image using the docker script. The build crashes during the first stage with the following error message:
```
W: Failure trying to run: chroot "/pi-gen/work/test/stage0/rootfs" /bin/true
W: See /pi-gen/work/test/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/test/stage0/rootfs/debootstrap': Directory not empty
[10:45:28] bootstrap failed: please check /pi-gen/work/test/stage0/debootstrap.log
```
and `tail - 5 /pi-gen/work/test/stage0/debootstrap.log`:
```
2022-02-16 10:45:22 URL:http://mirror.de.leaseweb.net/raspbian/raspbian/pool/main/v/vim/vim-common_8.2.2434-3+deb11u1_all.deb [226176/226176] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/vim-common_2%3a8.2.2434-3+deb11u1_all.deb" [1]
2022-02-16 10:45:23 URL:http://mirror.de.leaseweb.net/raspbian/raspbian/pool/main/v/vim/vim-tiny_8.2.2434-3+deb11u1_armhf.deb [600948/600948] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/vim-tiny_2%3a8.2.2434-3+deb11u1_armhf.deb" [1]
2022-02-16 10:45:24 URL:http://mirror.de.leaseweb.net/raspbian/raspbian/pool/main/v/vim/xxd_8.2.2434-3+deb11u1_armhf.deb [191608/191608] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/xxd_2%3a8.2.2434-3+deb11u1_armhf.deb" [1]
2022-02-16 10:45:26 URL:http://mirror.de.leaseweb.net/raspbian/raspbian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2_armhf.deb [84712/84712] -> "/pi-gen/work/test/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2_armhf.deb" [1]
chroot: failed to run command '/bin/true': Exec format error
```

Docker was complaining earlier in the build that: `The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested` and the error from debootstrap.log seems to point into a similar direction.

I am using all default values.
What can I do here?

XECDesign, 2022-02-16:
It looks like a binfmt issue, but I have no idea how Docker affects things, since that's maintained by the the community.

jnnks, 2022-02-16:
Yes, I agree. My distro does not come with a dedicated binfmt-support package and relies on systemd's implementation which was not accessible.

pirateradiohack, 2022-03-27:
hello @jnnks, excuse my ignorance, but how did you solve that? Thanks.

jnnks, 2022-03-27:
I ended up using a dedicated Ubuntu VM, since I was too lazy to figure out `systemd-binfmt.service` on Solus Linux :D
Maybe your distro is also relying on systemd?

pirateradiohack, 2022-03-27:
Thanks @jnnks for your answer and thank you for making me discover Solus Linux!

The system installed on my personal computer is Arch Linux. It's the first time I encounter this issue so I have not even dived yet into whether it's relying on systemd. 

Thanks for suggesting the VM solution, it will help temporarily. 

Finding a work around or a solution on a personal OS can be attained through tinkering. But I also rely on Continuous Integration services to automatically build the image and tinkering there is a bit more limited.

jnnks, 2022-03-28:
No problem at all, thanks for asking. I'm afraid I can't help you with Jenkins though, so good luck there :)

pirateradiohack, 2022-04-02:
It looks like building within the Ubuntu VM works (although I run into more issues, they seem to not be related to the actual build process). 
So in the end, I used to be able to build flawlessly on Travis-ci.com and my system (Arch Linux) with the Docker method. But now both the "normal" build.sh and the Docker method fail on both Travis and my system.

I had to resort to the solution above, installing a Ubuntu VM on top of Arch Linux. Then I use the Docker method. That's a lot of virtualisation / emulation / slicing to be able to build pi-gen. But for the moment I can confirm it seems to work fine.