RPi-Distro/pi-gen #593: How to configure mariadb inside script

gxbag, 2022-02-17:
Hi, I installed mariadb by adding it to `00-packages`, then I added the following between `on_chroot << EOF` and `EOF` in `00-run.sh`:

    systemctl stop mariadb
    systemctl start mariadb
    printf "CREATE DATABASE orders;\n" | mysql

This produces the following:

    Running in chroot, ignoring command 'stop'
    Running in chroot, ignoring command 'start'
    ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)

How can I configure mariadb within the script?

gxbag, 2022-02-18:
Thank you for your answer.

This could be helpful for recreating the database:

https://www.tecmint.com/transfer-mysql-databases-from-old-to-new-server/

I will try it.

XECDesign, 2022-02-21:
Unless I'm missing something, that just dumps the database and then re-creates it on the new server. You'll still run into the same issue where the mariadb server will need to be running in the chroot.