RPi-Distro/pi-gen #594: bootstrap failed

fheslouin, 2022-02-17:
Hey, 

My last build did not go through, I got this error : 
```
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot "/pi-gen/work/ideascube/stage0/rootfs" /bin/true
W: See /pi-gen/work/ideascube/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/ideascube/stage0/rootfs/debootstrap': Directory not empty
[16:10:45] bootstrap failed: please check /pi-gen/work/ideascube/stage0/debootstrap.log
```

Full log : https://paste.debian.net/1231259/

I'm using a fresh git cloned repo : `git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git`

XECDesign, 2022-02-21:
Hey, sorry it looks like your paste link expired.  What was the content of `debootstrap.log`?

I suspect this might be related: https://github.com/RPi-Distro/pi-gen/issues/590

fheslouin, 2022-02-21:
Yes, indeed same issue! I missed it out. Thanks

closed 2022-02-21 by fheslouin