RPi-Distro/pi-gen #595: 64 bits build on Pi4 running a 32 bits Raspberry OS Lite

gromain, 2022-02-20:
Hey,
I've been having issues with setting a 64 bits build with pi-gen on a Pi4 running a 32 bits Lite image.
The exact issue I'm seeing is a failure by `debootstrap` at the end :
```
[...]
2022-02-20 12:58:19 URL:http://deb.debian.org/debian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2_arm64.deb [87944/87944] -> "/home/rbazile/64bit-pi-gen/work/pi3-1-dev-64/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2_arm64.deb" [1]
chroot: failed to run command '/bin/true': Exec format error
```

I've installed both `qemu-user-static` and `binfmt-support`, binfmt is loaded in the kernel and `/proc/sys/fs/binfmt_misc` is filled:
```
rbazile@rbazile-dev:~/64bit-pi-gen $ ls /proc/sys/fs/binfmt_misc/
python3.9   qemu-cris  qemu-m68k        qemu-mips64    qemu-mipsn32    qemu-ppc64    qemu-riscv64  qemu-sh4eb        qemu-sparc64  qemu-xtensaeb
qemu-alpha  qemu-hppa  qemu-microblaze  qemu-mips64el  qemu-mipsn32el  qemu-ppc64le  qemu-s390x    qemu-sparc        qemu-x86_64   register
qemu-armeb  qemu-i386  qemu-mips        qemu-mipsel    qemu-ppc        qemu-riscv32  qemu-sh4      qemu-sparc32plus  qemu-xtensa   status
```

Is this expected behavior (not to be able to build the 64 bits version on the 32 bits OS) or am I missing something?
Thanks

XECDesign, 2022-02-21:
I haven't tried it but it 'should' work, in principle.

The main issue there is that there's nothing registered for aarch64 in that binfmt directory.

I can make chroot work on my pi zero by running:

```
echo ":test-arm64:M::\x7f\x45\x4c\x46\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-aarch64-static:OCF" | sudo tee /proc/sys/fs/binfmt_misc/register
```

However, since you're running it on a pi 4, it would be better to skip qemu entirely. If you put ` arm_64bit=1` in your `config.txt` and reboot, it will run with a 64-bit kernel. After that, it should "just work" (although I haven't tested this).

gromain, 2022-02-21:
I think that putting the `arm_64bit=1` did the trick. At least it managed to build the images.
I'll create a PR with a line in the doc for this.

XECDesign, 2022-02-21:
It's using it, just not for the bootstrap step. At the time I commented it out, debootstrap didn't work with it. Maybe whatever the issue was has been resolved and that line can be uncommented again.

gromain, 2022-02-21:
Ah yes, I'll test that and report then! :-)

gromain, 2022-02-21:
I didn't see any issue with removing the comment here https://github.com/RPi-Distro/pi-gen/blob/255288909beae446a4f1eeedaa544f1a7d421e54/scripts/common#L10
Build was nominal. I think this can be removed now.

PR #596 (2022-02-22): Provide information to build the 64 bit image in the README
Closes #595 and should help close other issues like #585.
commit b1d54c82 (2022-02-22): Provide information to build the 64 bit image in the README (#596)

Fixes #595
closed 2022-02-22 by XECDesign