RPi-Distro/pi-gen #597: Bullseye runs on CM4, buster does not run on CM4 - any idea?

capiman2, 2022-02-22:
We used pi-gen to build an image for a CM3 and it works since a long time.
Now we are doing first steps with CM4. We were able to generate an image with pi-gen and Bullseye for CM4.

We tried if the image for CM3 also works for CM4, but it does not seem to start.
I am even not sure, if it loads something at all from SD card. But I see LAN leds blinking, but no other reaction.
Putting CM4 on the same SD card works, I see log output from kernel over UART and can login via ssh.
("dtoverlay=disable-bt" was important, added it manually in both images, at least to get log output in Bullseye)

Has someone seen similar? Or has an explanation for this behaviour? 
Or "buster image" just not prepared for a CM4?

XECDesign, 2022-02-22:
I've just tested the official lite legacy image, which is built from the buster branch, and it seems to work.

Do you have a display connected? How are you determining that it's not working? What's the "other reaction" you're expecting?

What do you mean by "CM4 on the same SD card"?

capiman2, 2022-02-22:
"CM4 on the same SD card" -> wrong written by me, sorry. I meant "Bullseye on same SD card".

I am not able to connect a display. We have not connected the signals, because no use case.
It is a headless system.

Even when I remove the SD card, the LAN leds were still blinking. So it seems to be no sign that image has booted.
I am not sure, who controls the LEDs, if LAN controllers is doing it by his own, or CM4 is doing something with it.

Other reaction:  Like seeing log output on serial console. Or being able to connect via SSH.

I will also try the official lite legacy image.

XECDesign, 2022-02-22:
Since SSH and serial aren't enabled by default, it may be useful to know how you're enabling them. Do you have your images available anywhere?

Are you able to see the CM from the router by any chance (DHCP requests coming through, for example)?

> Even when I remove the SD card, the LAN leds were still blinking. So it seems to be no sign that image has booted.
> I am not sure, who controls the LEDs, if LAN controllers is doing it by his own, or CM4 is doing something with it.

@timg236 Are we expecting LAN LEDs to be active on a CM4 with no eMMC or SD card?

timg236, 2022-02-22:
"@timg236 Are we expecting LAN LEDs to be active on a CM4 with no eMMC or SD card?"

Not until some some software has initialised Ethernet e.g. Linux or the bootloader doing network boot.

XECDesign, 2022-02-22:
So I guess unless @capiman2 must have enabled network boot in the EEPROM, which makes it a little trickier to know whether Linux has booted.

capiman2, 2022-02-22:
The CM4 was given to me. I assume it was not used before.

I just have done:

xxx:~ $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

xxx:~ $ vcgencmd bootloader_version
Oct  2 2020 08:44:17
version bfe1c7ddc094b735134e96f2338c0ab2c9b10a5b (release)
timestamp 1601624657
update-time 0
capabilities 0x00000000
xxx:~ $

Does this help somehow? Or what can I use to query/enable/disable the network booting?
Or any other command so I see more, to verify it is starting or not.
In CM3 I think I was able to patch some files to enable UART output in bootloader.
But I think the way has changed for CM4. I am very new to CM4, so need more reading and experience.

XECDesign, 2022-02-22:
Doesn't look like network boot is enabled, so I'm not sure what would be flashing the LEDs.

Since the official image seems to be working, I don't think it's a pi-gen problem. Maybe it's better to address this on the forum?

timg236, 2022-02-22:
> So I guess unless @capiman2 must have enabled network boot in the EEPROM, which makes it a little trickier to know whether Linux has booted.

It's enabled by default in the manufacturing EEPROM and usbboot. Suggest adding BOOT_UART=1 in the EEPROM config and uart_2ndstage=1 in config.txt on the boot-media to get full UART tracing up to the Kernel start

capiman2, 2022-02-23:
I learned meanwhile how to update the EEPROM to update EEPROM content from 2020 to a brand new version of 2022.
And also enabled BOOT_UART=1.
Standard official buster image (legacy, 32bit) worked.

With our own image I see reproducable the following output:

`RPi: BOOTLOADER release VERSION:d8b208d3 DATE: 2022/02/22 TIME: 09:33:50 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1645522430 0xdfcfd361 0x00c03140 0x00064e1f
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200

RPi: BOOTLOADER release VERSION:d8b208d3 DATE: 2022/02/22 TIME: 09:33:50 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1645522430 0xdfcfd361 0x00c03140 0x00064796
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200

Boot mode: SD (01) order f2564
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
OCR c0ff8000 [163]
CID: 002750485344333247607cf261200131
CSD: 400e00325b590000e7bf7f800a400000
SD: bus-width: 4 spec: 2 SCR: 0x02058083 0x01000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  512000 type: 0x0c
MBR: 0x0007f000,49415355 type: 0x83
MBR: 0x02fa0000,10813440 type: 0x83
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 3939 c-count 504090 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 504090
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 3939 c-count 504090 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 504090
Read config.txt bytes     2324 hnd 0x00000000
Read start4.elf bytes  2239680 hnd 0x00000000
Read fixup4.dat bytes     5352 hnd 0x00000000
Firmware: 71bd3109023a0c8575585ba87cbb374d2eeb038f Dec  1 2021 15:01:54
0x00c03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+
`

After this with a working image, I see kernel booting. 
With a non-working buster image nothing more happens, even when waiting for over 2 minutes.

capiman2, 2022-02-23:
Don't know if it helps: My PC application tell me for kernel7l.img:

SHA256: 3DDBE3D25A4EB6B4B63BEEE42E01BEA8A53E679DBC802D7261CF21973AFC7530

timg236, 2022-02-23:
It looks like it's running the kernel then failing. Adding "earlycon=pl011,0xfe201000,115200n8" to cmdline.txt might reveal more.

The brfs File read lines indicate that there's a few dtoverlay files so commenting most of those out would simplify the boot behaviour.