RPi-Distro/pi-gen #600: APT Proxy fallback to direct on connection failure

umutediz, 2022-02-23:
We are generating images in an automated way using cron job and wrapper scripts and local APT proxy may be inaccessible from time to time due to some other external issues (network etc.). This causes the build script to fail obviously and interrupts the whole build process.

Build script already checks APT proxy connectivity before starting the actual build process but just exits with an error rather than ignoring APT proxy. I think adding a "fallback to direct connection" option via the config file (or as default behavior) might be useful in situations like unattended build processes. 

I can work on it if you think it'll be useful.

XECDesign, 2022-02-23:
I wouldn't mind if the fallback option was available, but not on by default.

umutediz, 2022-03-01:
I've made a PR related to this issue: #602 
Couldn't link to the issue.