RPi-Distro/pi-gen #601: Boot and root partitions are on different devices

caubut-charter, 2022-02-25:
```
branch: arm64
release: bullseye
boot medium: USB SSD (Samsung T7)
```

My images are not longer booting after commit https://github.com/RPi-Distro/pi-gen/commit/1ad53be314dbc73d7117dbd610a982c90e115f37.

It is triggering this error: https://github.com/RPi-Distro/raspi-config/blob/129b9461b3c068ee1a618fd3520f6c16afbcb530/usr/lib/raspi-config/init_resize.sh#L109

While troubleshooting, I did add a the DEV_NAMEs to the error message and one of them was a `loop` device instead of an `sdx` device.

Manually removing `.flush` from /etc/fstab allows the image to boot.

theGOTOguy, 2022-02-27:
I confirmed that removing `.flush` from [this line](https://github.com/RPi-Distro/pi-gen/blob/master/stage1/01-sys-tweaks/files/fstab#L2) fixes the issue.  Thank you for your research and the workaround, @caubut-charter.

XECDesign, 2022-02-28:
Should be fixed by https://github.com/RPi-Distro/pi-gen/commit/a120c4128ece237754af9e8bef6679d6a51f73ec

caubut-charter, 2022-03-03:
> Should be fixed by [a120c41](https://github.com/RPi-Distro/pi-gen/commit/a120c4128ece237754af9e8bef6679d6a51f73ec) (can't test it on my end right now)

Will test in the next couple hours.  Normally I netboot my image so fstab is re-written anyways, but I will be booting off USB SSD as soon as the image is done building.

caubut-charter, 2022-03-03:
Confirmed fixed.

closed 2022-03-03 by caubut-charter