RPi-Distro/pi-gen #605: en_US Locale

scottix, 2022-03-14:
I set the LOCALE to "en_US.UTF-8"

config
```
LOCALE_DEFAULT="en_US.UTF-8"
```

I get an error in the config

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory
```

I see similar error https://github.com/RPi-Distro/pi-gen/issues/65, but there was a fix implemented. Still getting it though.

XECDesign, 2022-03-22:
At what point are you seeing this?

After stage0/01-locale/00-packages runs, that message shouldn't come up.

If you're still seeing it after stage0/01-locale/00-packages, I think when chroot runs, it inherits the host system's locale. If you don't have it generated within the chroot, you'll see the warning. Note that it's not an error. It's just telling you that the locale is not available, so perl scripts are falling back to the "C" locale.

scottix, 2022-03-23:
It was present throughout all stages, I did see the fallback to C.

Everything seemed to work though, so maybe just a warning of such.

XECDesign, 2022-11-10:
Are you building with docker? If not, is host your system locale also set to en_US.UTF-8? If it's not, do you get the same error on your host system if you do set it to match?

closed 2024-04-11 by XECDesign