RPi-Distro/pi-gen #606: chroot: failed to run command 'dpkg-deb': No such file or directory

danielnixon, 2022-03-19:
* macOS Monterey 12.2.1, MacBook Pro 2016, Intel Core i7
* Vagrant 2.2.19
* VirtualBox 6.1.32
* pi-gen git commit: 5d044ba71b7eba33860fdbf222a9a29ae42691e9
* Vagrant image: ubuntu/xenial32 (also tried hashicorp/bionic64)
* Working in a new dir (/pi-builder), not the home dir

```
vagrant@vagrant:/pi-builder/pi-gen$ cat config 
IMG_NAME=test_dn
```

`sudo ./build.sh` output: 

```
vagrant@vagrant:/pi-builder/pi-gen$ sudo ./build.sh
[23:57:14] Begin /pi-builder/pi-gen
[23:57:14] Begin /pi-builder/pi-gen/stage0
[23:57:14] Begin /pi-builder/pi-gen/stage0/prerun.sh
[23:57:14] End /pi-builder/pi-gen/stage0/prerun.sh
[23:57:14] Begin /pi-builder/pi-gen/stage0/00-configure-apt
[23:57:14] Begin /pi-builder/pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/pi-builder/pi-gen/work/test_dn/stage0/rootfs/etc/apt/sources.list.d/' is not a directory: No such file or directory
```

`sudo CLEAN=1 ./build.sh |& tee build.log` output (snipped):

```
I: Extracting libudev1...
I: Extracting libuuid1...
I: Extracting libzstd1...
I: Extracting login...
I: Extracting logsave...
I: Extracting lsb-base...
I: Extracting mawk...
I: Extracting mount...
I: Extracting ncurses-base...
I: Extracting ncurses-bin...
I: Extracting passwd...
I: Extracting perl-base...
I: Extracting sed...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot /pi-builder/pi-gen/work/test_dn/stage0/rootfs dpkg-deb -f /var/cache/apt/archives/dpkg_1.20.9+rpi1_armhf.deb Version
W: See /pi-builder/pi-gen/work/test_dn/stage0/rootfs/debootstrap/debootstrap.log for details
W: Failure trying to run: chroot /pi-builder/pi-gen/work/test_dn/stage0/rootfs mount -t proc proc /proc
W: See /pi-builder/pi-gen/work/test_dn/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-builder/pi-gen/work/test_dn/stage0/rootfs/debootstrap': Directory not empty
[00:03:11] bootstrap failed: please check /pi-builder/pi-gen/work/test_dn/stage0/debootstrap.log
vagrant@vagrant:/pi-builder/pi-gen$ cat /pi-builder/pi-gen/work/test_dn/stage0/debootstrap.log
gpgv: Signature made Fri Mar 18 22:35:03 2022 UTC
gpgv:                using RSA key A0DA38D0D76E8B5D638872819165938D90FDDD2E
gpgv: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
chroot: failed to run command 'dpkg-deb': No such file or directory
chroot: failed to run command 'mount': No such file or directory
vagrant@vagrant:/pi-builder/pi-gen$
```

Am I doing something obviously wrong? Should I give up and run on real arm hardware?

XECDesign, 2022-03-21:
Hmm sorry, I don't think I've seen this one before and it's not the kind of setup I'd use myself. It's something I'd need to have in front of me to be able to debug.

Kudesnick, 2022-03-30:
I faced the same problem. on different host machines: WSL2 Ubuntu 18.04/20.04. I used oficial microsoft kernel and custom (see https://www.frakkingsweet.com/running-pi-gen-on-wsl2/). Right now I'm using a cloud server with Ubuntu 18.04 and I'm seeing the same problem.

`
I: Extracting zlib1g...
W: Failure trying to run: chroot /home/user/pi-gen/work/CrazyPi/stage0/rootfs dpkg-deb -f /var/cache/apt/archives/dpkg_1.20.9+rpi1_armhf.deb Version
W: See /home/user/pi-gen/work/CrazyPi/stage0/rootfs/debootstrap/debootstrap.log for details
W: Failure trying to run: chroot /home/user/pi-gen/work/CrazyPi/stage0/rootfs mount -t proc proc /proc
W: See /home/user/pi-gen/work/CrazyPi/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/home/user/pi-gen/work/CrazyPi/stage0/rootfs/debootstrap': Directory not empty
[22:16:08] bootstrap failed: please check /home/user/pi-gen/work/CrazyPi/stage0/debootstrap.log
user@00-00-00-00:~/pi-gen$ 
`
debootstrap.log:
`
gpgv: Signature made Wed Mar 30 19:32:41 2022 MSK
gpgv: using RSA key A0DA38D0D76E8B5D638872819165938D90FDDD2E
gpgv: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
chroot: failed to run command 'dpkg-deb': No such file or directory
chroot: failed to run command 'mount': No such file or directory
`
The file `/var/cache/apt/archives/dpkg_1.20.9+rpi1_armhf.deb` is on the disk.

nicolas17, 2022-08-03:
I'm getting the same problem on a xenial32 VM. Turns out when the system tries to run any ARM binary inside the chroot, it looks for /usr/bin/qemu-arm-static inside the chroot too, so qemu-arm-static needs to be copied into the rootfs before even running debootstrap.

I don't know why I didn't hit this in debian bullseye amd64, maybe either debootstrap or binfmt works differently there...

XECDesign, 2022-08-04:
Yeah I think I know what's happening. Older versions of qemu-static didn't pass the 'F' flag to binfmt and I guess that's the case for xenial.

barbolo, 2022-08-04:
@XECDesign what version of Ubuntu would you recommend? Is the statement below in README.md still valid?

`Currently it is only supported on either Debian Buster or Ubuntu Xenial and is known to have issues building on earlier releases of these systems.`

barbolo, 2022-08-04:
I'm using multipass (driver: Vagrant) in macOS running Ubuntu 20.04. It is finally building RPI images.

It wasn't working with build-docker, neither with Ubuntu 22.04 or 16.04 (xenial).