RPi-Distro/pi-gen #610: install: target '/pi-gen/work/2022-04-15-Raspbian/stage0/rootfs/etc/apt/' is not a directory: No such file or directory

jerrychong25, 2022-04-15:
Hi all,

I'm facing some compilation issue on generating own Raspbian Image.

After run build command, apparently it hits following error:

`install: target '/pi-gen/work/2022-04-15-Raspbian_LinkUp/stage0/rootfs/etc/apt/' is not a directory: No such file or directory`

Here is the error screenshot:
[image]

Environment:
- macBook Pro 2021 14 inch Apple M1 Pro Chip
- macOS Monterey

Is there any alternative/solution to make it work on this environment? 

If not, what is the best alternative solution (Ubuntu/Debian) to build the same image?

Thanks in advance!

XECDesign, 2022-04-19:
That's not an environment I'd expect things to work on. I don't have access to the appropriate hardware to investigate how to make it work.

I would recommend building natively on a pi under Raspberry Pi OS with a sufficiently large storage medium.

jerrychong25, 2022-04-20:
> That's not an environment I'd expect things to work on. I don't have access to the appropriate hardware to investigate how to make it work.
> 
> I would recommend building natively on a pi under Raspberry Pi OS with a sufficiently large storage medium.

Hi @XECDesign ,

Noted, I will try again on my end for Raspberry Pi 4 directly.

jerrychong25, 2022-04-28:
Managed to build same code in Pi 4 Firmware v1.5, will close this issue

Thanks @XECDesign !

closed 2022-04-28 by jerrychong25

thk686, 2025-03-31:
I have just encountered this using build-docker.sh on osx.