RPi-Distro/pi-gen #614: First User creation with new first-boot wizard

thelouisvivier, 2022-04-28:
Hey people !

Since 04/04, the default "pi" user has been removed. Now, a first-boot wizard enforces the creation of a new user. 

Following that, it seems that the config variable FIRST_USER is not working anymore. So, even with that variable set, the first-boot wizard is opened. 

How can I create a first user, without that wizard being opened ? I would like to not create a first user, and deploy headless RPIs with images running out of the box.

Thanks for your help :)

XECDesign, 2022-05-03:
You could create images with a userconf.txt file on the boot partition.

I haven't tested this, but you could also try removing `export-image/01-user-rename`. We could add a variable to determine whether or not `export-image/01-user-rename/01-run.sh` should run.

XECDesign, 2022-05-04:
That's not different from any other way the mechanism runs. We will restore the previous behaviour where by default the lite image does not autologin in the next image.

It happens [here](https://github.com/RPi-Distro/userconf-pi/blob/master/cancel-rename#L28=)

XECDesign, 2022-05-04:
It should be changed soon.

thelouisvivier, 2022-05-04:
The thing is that even with a created user (via $FIRST_USER_NAME), the first boot wizard was launched...
Creating a user with userconf on the boot partition was not possible for me, as I wanted to configure that user before the first boot. 

I finally could get rid of that wizard by disabling the userconf process during the build process.

andi34, 2022-05-13:
Hey there!
I've got the same problem the defined user was not setup on boot. 
My config:
https://github.com/andi34/photobooth/blob/dev/image_builder/raspberry/custom-pi-config

Since I am using a GitHub action ATM, would it be enough to delete `export-image/01-user-rename`  before the build runs?

thelouisvivier, 2022-05-13:
Hey,

To revert to the old way I did the following :

- still using FIRST_USER_NAME var to create a first user
- disable users creation on first boot by adding theses lines in a stage 01-run.sh :

````

# Disable raspiconfig at boot
on_chroot << EOF
if [ -e /etc/profile.d/raspi-config.sh ]; then
                rm -f /etc/profile.d/raspi-config.sh
                if [ -e /etc/systemd/system/getty@tty1.service.d/raspi-config-override.conf ]; then
                               rm /etc/systemd/system/getty@tty1.service.d/raspi-config-override.conf
                fi
                telinit q
fi
EOF
 
# Disable autologin
on_chroot << EOF
systemctl --quiet set-default multi-user.target
rm -f /etc/systemd/system/getty@tty1.service.d/autologin.conf
EOF
 
#Disable users creation on first boot
on_chroot << EOF
systemctl stop userconfig
systemctl disable userconfig
systemctl mask userconfig
EOF

````

Feel free to ask any questions on that configuration !

Hope it helps :)

andi34, 2022-05-13:
Great! Thanks a lot. I'll add this to https://github.com/andi34/photobooth/tree/dev/image_builder/raspberry/stage-photobooth

andi34, 2022-05-13:
But raspi-config at boot and autologin could be kept, right? Or do they stick together with the new user creation?

andi34, 2022-05-13:
Is `01-run.sh` executed by default? Or do I need to add something like `stage-photobooth/00-disable-usercreation/00-run.sh`? `00-run.sh` was only mentioned in the README.
`stage-photobooth` was defined to be included.

Sorry if it's a stupid question, but GitHub action takes >1 hour and would be great to save some time trying.

andi34, 2022-05-14:
[image]

We'll, still getting the first start wizard which needs a new user to be added... if I type the same username and password as given by my config it won't execute the Autostart I've added before.

andi34, 2022-05-16:
I didn't found a proper solution to restore the previous behavior since the first start wizard forces me to add a new username.
I wonder if deleting `/etc/xdg/autostart/piwiz.desktop`  will help, but sadly wouldn't allow to use the other funtionalitys of the Pi Wizard.
I am installing some extra programs and adding some extra scripts for the default user I am setting up in de build config. Those userspecific changes get lost while the user get renamed/created.

For reference affected parts of my installer which runs in a separate stage at the end:

https://github.com/andi34/photobooth/blob/a9255bd23bdb77b73b07903e2d80980240ad9b90/install-raspbian.sh#L514-L559

Any other idea to workaround the user creation / improve my setup?

thelouisvivier, 2022-05-16:
When I was searching for a solution, I found this [thread](https://forums.raspberrypi.com/viewtopic.php?t=231557). 
You should look into it, it's about deleting the script you mentioned before

andi34, 2022-05-16:
Thanks for your answer. Yeah, I found that thread too, but it's not the best solution. There should be an option to only disable/skip the user creation if one is setup by default imo.

andi34, 2022-05-17:
This doesn't help, sadly...
I've also tried building an image using Pi OS Desktop x86 setup on Virtual Box. The setup wizard is still there and even comes back after a reboot... Like the setup was never running.
If I use ctr+alt+del to logout I can not see the new user name I've added before...

And another problem: if I now login with the user I've set while building the image, there's no start menu bar visible...

[image]

[image]

Any idea of what's wrong?
My custom changes to the build process:
https://github.com/andi34/photobooth/tree/dev/image_builder/raspberry/stage-photobooth

andi34, 2022-05-19:
To answer to myself:

Deleting `/etc/xdg/autostart/piwiz.desktop` won't do anything in my custom stage because export-image/01-user-rename is executed afterwards.

`/etc/xdg/autostart/piwiz.desktop` also gets created afterwards https://github.com/RPi-Distro/userconf-pi/blob/5bc5c03f9bafe8300119e8be3829c69abcada0f4/rename-user#L50-L56

The only option seems to be to remove or modify the `01-run.sh` as mentioned by @XECDesign .
I'd be happy if there's a flag instead to disable this behavior or allow to skip the user rename/creation from PiWiz if a valid user exists already. This way there wouldn't be the need to change files in this repo which would make an local update of this repo easier (sure, I can discard or commit the local changes first, but I prefer to keep git tracked files untouched).

A new build is compiling right now. I'll report back.

About the missing menu bar I found this
https://www.kirkmakse.com/raspberry-pi-raspbian-missing-menu-bar/

I'll check on the new build and test if the menu bar is still missing.

gromain, 2022-05-20:
I agree this change definitely breaks a lot of use cases (and I mean, a lot. On my side it's 3 images that are rendered unusable by this change).

gromain, 2022-05-20:
For users that don't have lightdm installed in their image, only the SSH banner will be displayed.

One simple fix for this is simply to remove this folder `export-image/01-user-rename/`.

@XECDesign would you agree on having the `on_chroot` part of the [export-image/01-user-rename/01-run.sh script](https://github.com/RPi-Distro/pi-gen/blob/master/export-image/01-user-rename/01-run.sh) in a conditional block that checks if `$FIRST_USER_NAME` is equal to `pi` and only executes this if that's the case ?

Also, just to understand, what was the rationale for having this step in the export-image stage instead of in another (regular) stage?

andi34, 2022-05-20:
@gromain maybe because it's executed definitely and a core security reason. I'd also not check for a specific username, I'd suggest to add a flag to skip this.

gromain, 2022-05-20:
@andi34 I agree it's a security feature, but protecting people using the official image mainly.

If you create your image with a set `FIRST_USER_NAME`, it means that you want another username for your first user in your image. Why then force the creation of a new user on first boot? There is no point in having `FIRST_USER_NAME` in the config file in that case.

IMO, if `FIRST_USER_NAME` is set in config (and set to something else than the default `pi` used during the image creation), then it makes sense this user is found in the final image, without having an additional flag.

Maybe we could print a warning informing the user of the risks in this case? Something like "You chose to create a default username for your image, please be aware of the security implications see xyz.com". I feel like if you choose to create a custom image, and you choose to change the default username, you can be trusted with managing the security of your device.

XECDesign, 2022-05-20:
> @XECDesign would you agree on having the `on_chroot` part of the [export-image/01-user-rename/01-run.sh [...]

I think there's an argument that shipping images with a default username and password in general is a bad practice. So if you want to force it, it should be an extra configuration option, not just a username check.

> Also, just to understand, what was the rationale for having this step in the export-image stage instead of in another (regular) stage?

The behaviour of the script is different depending on whether it runs on a system with a desktop environment or not. If we were to run it in stage 2, for example, then it won't work properly in stage 4. It would need to run twice, so it just made more sense to me to do it at the end, once the environment is already determined.

gromain, 2022-05-20:
> I think there's an argument that shipping images with a default username and password in general is a bad practice. So if you want to force it, it should be an extra configuration option, not just a username check.

I agree it's bad practice and I agree with having this set by default for the `pi` user and most users that don't customize their image.

But I believe in this case setting `FIRST_USER_NAME` and `FIRST_USER_PASSWORD` is this extra option. As I said, maybe we can just have an extra warning at the beginning of the build that this configuration is not recommended if the image is distributed, but trust the `pi-gen` user to behave responsibly. Or let's just remove `FIRST_USER_NAME` from config, because I think this will lead to more confusion if you have to setup `FIRST_USER_NAME` and `FORCE_USER` (setting name not definitive!).

For example, I use pi-gen to generate images for my own direct use. I don't understand why I should be forced to set something else manually on the first boot in addition to `FIRST_USER_NAME` and `FIRST_USER_PASSWORD`. Those are already the steps I take to explicitly tell pi-gen to create a user with a username and password I chose.

Another example, for an embedded Raspberry Pi system where you use `pi-gen` to create your image, forcing the user to be created on the first boot is less than ideal since having someone to input a username on the first boot may not be an option for this use case.

> The behaviour of the script is different depending on whether it runs on a system with a desktop environment or not. If we were to run it in stage 2, for example, then it won't work properly in [...]

Ok, the decision to put it in `export-image` makes sense to me now. I found it weird because I feel this step is directly related to the OS you are generating, and is not specific to the step of exporting an image, hence my thinking to put it in a "normal" stage. But indeed, it applies to several images at once.

andi34, 2022-05-20:
> To answer to myself:
> 
> Deleting `/etc/xdg/autostart/piwiz.desktop` won't do anything in my custom stage because `export-image/01-user-rename` is executed afterwards.
> 
> `/etc/xdg/autostart/piwiz.desktop` also only gets created afterwards https://github.com/RPi-Distro/userconf-pi/blob/5bc5c03f9bafe8300119e8be3829c69abcada0f4/rename-user#L50-L56
> 
> The only option seems to be to remove or modify the `01-run.sh` as mentioned by @XECDesign . I'd be happy if there's a flag instead to disable this behavior or allow to skip the user rename/creation from PiWiz if a valid user exists already. This way there wouldn't be the need to change files in this repo which would make an local update of this repo easier (sure, I can discard or commit the local changes first, but I prefer to keep git tracked files untouched).
> 
> A new build is compiling right now. I'll report back.
> 

Works :) but still the flag would be my preferred way to not conflict with the repo on update.

> About the missing menu bar I found this https://www.kirkmakse.com/raspberry-pi-raspbian-missing-menu-bar/
> 
> I'll check on the new build and test if the menu bar is still missing.

Working now like it should. Seems like I've messed the permissions on the .config folder with my photobooth installation script (on an already booted image the .config folder exists already while it does not generating the image).

XECDesign, 2022-05-20:
> user to behave responsibly

I've been burned by that assumption too many times. While it's quite redundant, I think it would need to be an option like `DISABLE_USER_RENAME` in addition to the existing username and password settings.

KwadFan, 2022-05-21:
Would it be possible to make something like /boot/ssh file for Images, that uses for example CustomPIOS to manipulate existing images, also?

andi34, 2022-05-21:
There's a flag already for it to enable ssh by default using pi-gen for custom images
> `ENABLE_SSH` (Default: `0`)
>
> Setting to `1` will enable ssh server for remote log in. > Note that if you are using a common password such as the defaults there is a high risk of attackers taking over you Raspberry Pi.

If there's a different project used to build the custom image you should reach out to the developer (s) of this custom project.

Without much trouble you can add an empty file called `ssh` to the boot partition of PiOS which can be accessed on Linux and Windows easily.

But guess this is OT here.

KwadFan, 2022-05-21:
> There's a flag already for it to enable ssh by default using pi-gen for custom images
> 
> > `ENABLE_SSH` (Default: `0`)
> > Setting to `1` will enable ssh server for remote log in. > Note that if you are using a common password such as the defaults there is a high risk of attackers taking over you Raspberry Pi.
> 
> If there's a different project used to build the custom image you should reach out to the developer (s) of this custom project.
> 
> Without much trouble you can add an empty file called `ssh` to the boot partition of PiOS which can be accessed on Linux and Windows easily.
> 
> But guess this is OT here.
> 
> Edit: To answer your question:
> 
> 1. Make a bootable stick/sdcard with the image.
> 2. Mount the boot partition and create the `ssh` file.
> 3. On Linux use the `dd` command line to create an iso from your modified stick/sdcard, on Windows e.g. use Win32-Disk-Imager.

I think, I didn't use the right words or explainations.
Yes, I know that I can simply add a empty `ssh` file.

What I was thinking of, is there a possibilty to add a routine looking for a, for ex., `disable-rename` file to skip the whole process of renaming and disabling userconf-pi service at a time?

Regards

andi34, 2022-05-21:
@KwadFan that make sense. Well, maybe this is a good idea because we don't need to skip the whole PiWiz process, only the user rename part of it.
Things like setting up WiFi, setting location and keyboard are sure useful. So the file should be generated using pi-gen and PiWiz should check for it?

gromain, 2022-06-13:
To move this issue forward @XECDesign, would you agree on having a flag in `config` to force the creation of a username and password as defined by `FIRST_USER_NAME` and `FIRST_USER_PASS` ?
I suggest `FORCE_USER_CREATION` as `DISABLE_USER_RENAME` is not very clear. One has to know already that the chosen username/password will be removed (actually, renamed) during the first boot. `FORCE_USER_CREATION` is clearer on what is happening. The Readme will also reflect this change.

[export-image/01-user-rename/01-run.sh script](https://github.com/RPi-Distro/pi-gen/blob/master/export-image/01-user-rename/01-run.sh) would be as such:
```
#!/bin/bash -e

if [ "${FORCE_USER_CREATION}" ]; then
	on_chroot << EOF
		SUDO_USER="${FIRST_USER_NAME}" rename-user -f -s
EOF
fi
```

closed 2022-06-17 by XECDesign

andi34, 2022-09-11:
@XECDesign PiWiz still likes to rename the existing user. There's a way to skip that on PiWiz?
If I purge PiWiz I am getting issues with the network not being set up 🥴

andi34, 2022-09-11:
Ok, I've got an answer on piwiz repo.

> I'm afraid it is not possible to skip user creation in the wizard.
> 
> You can, however, preconfigure a user account when the card is flashed using Raspberry Pi Imager, or by putting a configuration file in the boot partition of a flashed image, as described on this page: https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

does mean skipping it here makes no sense as I am still forced to create a new user...

XECDesign, 2022-09-22:
I think we need to add an extra step of removing `/etc/xdg/autostart/piwiz.desktop` if user rename is disabled. It does mean you lose the first run wizard altogether, but it wouldn't work properly with this setup anyway.

XECDesign, 2022-09-22:
Just pushed a commit that should do that.

andi34, 2022-09-22:
Thanks a lot @XECDesign !
On a test image I've purged `piwiz` from the build, but network was not working on that image. Does PiWiz do some magic on that? If yes, do we loose that magic?

XECDesign, 2022-09-22:
Shouldn't be any network-related magic from piwiz, no.

guysoft, 2025-03-26:
> It should be changed soon.

H3y @XECDesign I just had a user open this is happening in an image from ``2025-01-28``:
https://github.com/guysoft/FullPageOS/issues/639
Could this be a regression where the auto login tty is logging in again?

XECDesign, 2025-04-23:
@guysoft things seem to work as expected on the PiOS end, AFAICT.

If the image in question has lightdm with autologin enabled, then I think it would take the 'B4' path.

guysoft, 2025-07-20:
Hey, lookin at this again.
It does take the B4 path.
The issue is that users are asking for the UI to autologin, and for the tty to not autologin. Is there a way to do taht with B4?

For context, this is the UI menu. I will convert this to scripting but it seems like its a limitation in raspi-config too.
[image]

Without this option it means there is a issue that the kiosks have a potential security hole.

XECDesign, 2025-07-21:
> The issue is that users are asking for the UI to autologin, and for the tty to not autologin. Is there a way to do taht with B4?

I don't think so. It would need to be an additional step of either disabling tty1 autologin or enabling lightdm autologin.

Maybe raspi-config should split desktop and tty1 autologin into two separate nonint options. For now I'd just recommend removing the tty1 autologin drop-in file or using rpi-image-gen instead (which comes with a kiosk config example).

guysoft, 2025-07-21:
Where does the auto tty login happen?
Someone mentioned ``getty@tty1``, but that does not seem to be in the image, is it generated after the first boot?

PR #618 (2022-06-17): Closes #614 by creating a `FORCE_USER_CREATION` flag to be set in config
This deactivates the export-image/01-user-rename/01-run.sh script when the `DISABLE_FIRST_BOOT_USER_RENAME` flag is set to `1`.
Closes #614.
commit 01b24320 (2022-06-17): Create a `DISABLE_FIRST_BOOT_USER_RENAME` flag to be set in config (#618)

Closes #614