RPi-Distro/pi-gen #619: Building pi-gen with vagrant

pirateradiohack, 2022-06-26:
I have used different methods to build pi-gen from my workstation (Arch Linux, intel 64bits). But as of late, even the docker method is failing for me.
I was recommended in another issue to try a virtual machine with an OS listed in the supported list: Debian Buster or Ubuntu Xenial.
I tried and it seems to work. In order to streamline the experience I find vagrant quite straightforward.

Here is my Vagrantfile for anyone interested:

```
Vagrant.configure("2") do |config|
  config.vm.box = "debian/contrib-buster64"

  config.vm.synced_folder ".", "/pigen"

  config.vm.provider "virtualbox" do |vb|
    vb.memory = "4096"
  end
  config.vm.provision "shell", inline: <<-SHELL
    apt-get update
    apt-get install -y coreutils quilt parted qemu-user-static debootstrap zerofree zip \
    dosfstools libarchive-tools libcap2-bin grep rsync xz-utils file git curl bc \
    qemu-utils kpartx gpg pigz
  SHELL
end
```
You'll need to install vagrant and virtualbox and then do `vagrant up` and `vagrant ssh`.

When this file is in the pi-gen repo it shares the folder in `/pigen` so you can just `cd /pigen` and `sudo ./build.sh` and get your images in deploy in your local filesystem.

If you would like me to do a PR with that file and the instructions on how to use it in the README just let me know and I'll take care of that. If not, feel free to close the issue.

XECDesign, 2022-06-29:
I'm not sure. Vagrant is definitely a much better option than Docker, but I don't want to add 100 different containerisation and virtualisation alternatives I don't use and can't support.

pirateradiohack, 2022-06-29:
Sure, that's why I did not do a PR straight away. 

Let's leave that in the ticket then, and I will update it if I find some improvements and will help if I can.

Meanwhile I'll go ahead and close it.

closed 2022-06-29 by pirateradiohack