RPi-Distro/pi-gen #620: failed to write (No space left on device)

cdupont, 2022-07-08:
I get the following error:
```
Preparing to unpack .../184-pocketsphinx-en-us_0.8+5prealpha+1-13_all.deb ...
Unpacking pocketsphinx-en-us (0.8+5prealpha+1-13) ...
dpkg: error processing archive /tmp/apt-dpkg-install-DgcFA0/184-pocketsphinx-en-us_0.8+5prealpha+1-13_all.deb (--unpack):
 cannot copy extracted data for './usr/share/pocketsphinx/model/en-us/en-us/mdef' to '/usr/share/pocketsphinx/model/en-us/en-us/mdef.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../185-raspi-config_20220506_all.deb ...
Unpacking raspi-config (20220506) over (20220301) ...
Preparing to unpack .../186-rpi-eeprom_13.14-1_arm64.deb ...
Unpacking rpi-eeprom (13.14-1) over (13.10-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-DgcFA0/186-rpi-eeprom_13.14-1_arm64.deb (--unpack):
 cannot copy extracted data for './lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-04-16.bin' to '/lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-04-16.bin.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

I'm not sure what it means.
Here is the result of `df`:
```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7,7G     0  7,7G   0% /dev
tmpfs           1,6G  2,6M  1,6G   1% /run
/dev/nvme0n1p2  468G  186G  259G  42% /
tmpfs           7,7G     0  7,7G   0% /dev/shm
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           7,7G     0  7,7G   0% /sys/fs/cgroup
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop1       56M   56M     0 100% /snap/core18/2344
/dev/loop3       55M   55M     0 100% /snap/snap-store/558
/dev/loop2      219M  219M     0 100% /snap/gnome-3-34-1804/72
/dev/loop4      4,8M  4,8M     0 100% /snap/yq/1792
/dev/loop5       62M   62M     0 100% /snap/core20/1494
/dev/loop6       47M   47M     0 100% /snap/snapd/16010
/dev/loop7       51M   51M     0 100% /snap/snap-store/547
/dev/nvme0n1p1  511M  7,9M  504M   2% /boot/efi
/dev/loop8      4,8M  4,8M     0 100% /snap/yq/1787
/dev/loop9      219M  219M     0 100% /snap/gnome-3-34-1804/77
/dev/loop10      62M   62M     0 100% /snap/core20/1518
/dev/loop11      82M   82M     0 100% /snap/gtk-common-themes/1534
/dev/loop12      47M   47M     0 100% /snap/snapd/16292
/dev/loop13     255M  255M     0 100% /snap/gnome-3-38-2004/106
/dev/loop14      56M   56M     0 100% /snap/core18/2409
/dev/loop15      92M   92M     0 100% /snap/gtk-common-themes/1535
/dev/loop16     401M  401M     0 100% /snap/gnome-3-38-2004/112
tmpfs           1,6G   36K  1,6G   1% /run/user/1000
tmpfs           1,6G  8,0K  1,6G   1% /run/user/126
/dev/sda2       2,9G  2,3G  483M  83% /media/cdupont/rootfs
/dev/sda1       253M   31M  222M  12% /media/cdupont/boot
/dev/loop18     2,9G  2,9G     0 100% /var/lib/jenkins/workspace/5_ISO/work/WaziGate/export-image/rootfs
/dev/loop17     253M   27K  253M   1% /var/lib/jenkins/workspace/5_ISO/work/WaziGate/export-image/rootfs/boot
```

XECDesign, 2022-07-08:
At which stage are you adding these new files and what's the filesystem type of `/dev/nvme0n1p2`? It's looking like things got filled up during the export stage, but at that point, there shouldn't be new packages being installed.

cdupont, 2022-07-08:
The files are installed at stage 3.
/dev/nvme0n1p2 is ext4.

The stage 3 script is here: https://github.com/Waziup/wazigate-gen/blob/arm64-dev/stage3/03-wazigate/00-run.sh
Thank you so much!

XECDesign, 2022-07-08:
Okay, that's not what I was expecting and I don't think I understand what could be happening.

The output you've shown fails in stage3, but you're showing something the export-image stage actually being full.

Unless you're using `USE_QCOW2`, the files should just be being stored on wherever you're running pi-gen from. Are you running it through Docker or is this a straight build with no containers running from your root partition?

cdupont, 2022-07-08:
It is built by a Jenkins script: https://github.com/Waziup/wazigate-gen/blob/arm64-dev/Jenkinsfile
We don't use `USE_QCOW2`. There is no docker involved (as far as I know)...

XECDesign, 2022-07-08:
I'm not seeing any obvious issues. Maybe adding a 'df' in a run.sh file somewhere right before it fails could reveal something we're missing?

cdupont, 2022-07-08:
I placed a `df -h` at the end of run.sh of stage 3:
```
Filesystem      Size  Used Avail Use% Mounted on
udev            7,7G     0  7,7G   0% /dev
tmpfs           1,6G  2,6M  1,6G   1% /run
/dev/nvme0n1p2  468G  186G  259G  42% /
tmpfs           7,7G     0  7,7G   0% /dev/shm
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           7,7G     0  7,7G   0% /sys/fs/cgroup
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop1       56M   56M     0 100% /snap/core18/2344
/dev/loop3       55M   55M     0 100% /snap/snap-store/558
/dev/loop2      219M  219M     0 100% /snap/gnome-3-34-1804/72
/dev/loop4      4,8M  4,8M     0 100% /snap/yq/1792
/dev/loop5       62M   62M     0 100% /snap/core20/1494
/dev/loop6       47M   47M     0 100% /snap/snapd/16010
/dev/loop7       51M   51M     0 100% /snap/snap-store/547
/dev/nvme0n1p1  511M  7,9M  504M   2% /boot/efi
/dev/loop8      4,8M  4,8M     0 100% /snap/yq/1787
/dev/loop9      219M  219M     0 100% /snap/gnome-3-34-1804/77
/dev/loop10      62M   62M     0 100% /snap/core20/1518
/dev/loop11      82M   82M     0 100% /snap/gtk-common-themes/1534
/dev/loop12      47M   47M     0 100% /snap/snapd/16292
/dev/loop13     255M  255M     0 100% /snap/gnome-3-38-2004/106
/dev/loop14      56M   56M     0 100% /snap/core18/2409
/dev/loop15      92M   92M     0 100% /snap/gtk-common-themes/1535
/dev/loop16     401M  401M     0 100% /snap/gnome-3-38-2004/112
tmpfs           1,6G   36K  1,6G   1% /run/user/1000
tmpfs           1,6G  8,0K  1,6G   1% /run/user/126
/dev/sda2       2,9G  2,3G  483M  83% /media/cdupont/rootfs
/dev/sda1       253M   31M  222M  12% /media/cdupont/boot
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/2765656ace87d673a225e1b3c6251f4f907329a88a18eef6bd299e35fb8cbb12/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/9fca571ad47924dfbc34138854c286117b9487806ac857d2154ae41be96ef004/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/fba5e484bdf4a9e79906e555e60728fca301fb1ad088aa4d36794db26fa6b8f2/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/4da6648650f4a9f01d2407ffe6cb8c75cb3f2ac56ba3f87e3afb9031d4e4f2b9/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/33ffdec506e7e02a5e15bf4a5fe12ceab6def2b9ffe06bcafd36423eeb2bc6a3/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/e977669a307ce4b71dc729acfe3763e11259af9427e57a7f6a1fb4b8938f054f/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/590362e0ec19c5d5faa2b8743586603542e4fe3fe1c2e415ccc7cd2f6f6ad786/merged
overlay         468G  186G  259G  42% /var/lib/docker/overlay2/d90827afd0a4f997f5655332f433db8c7d7f06faae54a2ed8afb35366c85b02b/merged
/dev/loop18     2,9G  2,9G     0 100% /var/lib/jenkins/workspace/5_ISO/work/WaziGate/export-image/rootfs
/dev/loop17     253M   27K  253M   1% /var/lib/jenkins/workspace/5_ISO/work/WaziGate/export-image/rootfs/boot
```

XECDesign, 2022-07-11:
Sorry, don't think I can be of any help on this one. I suspect that there are some docker or tmpfs shenanigans going on in jenkins, but I have no idea.