RPi-Distro/pi-gen #629: raspbian-archive-keyring package

ghadf, 2022-08-27:
I have been experimenting with pi-gen and from what I can see, raspbian-archive-keyring installed during debootstrap invocation (a) uses the now deprecated apt-key and (b) has a dependency on gnupg (rather than just gpg?), which in turn pulls in many more dependencies.

Maybe I've got the wrong end of the stick but can't/shouldn't raspbian-archive-keyring be updated to add the key to /etc/apt/trusted.gpg.d and remove the dependency?

(Previously raised at https://forums.raspberrypi.com/viewtopic.php?t=339417 and directed here.)

XECDesign, 2022-08-30:
Maybe, but raspbian-archive-keyring isn't one of our packages. That comes directly from Raspbian.

https://bugs.launchpad.net/raspbian

closed 2022-08-30 by XECDesign

ghadf, 2022-08-31:
For reference, the already-existing bug report I missed: https://bugs.launchpad.net/raspbian/+bug/1727874