RPi-Distro/pi-gen #630: Question: do you have a recommended way to test the generated images?

pirateradiohack, 2022-09-06:
I would like to test my images once they have built. I guess it's a pretty common practice and I wonder how it is usually done.

Ideally I would like to automatically test, for instance, if the user that was supposed to be created by the build is actually present on the resulting image. If the software that was supposedly installed is actually there and executable and behaves predictably. That kind of things. 

I guess anything related to hardware must be tested on the device itself and not within Qemu or anything that can be launched automatically during a CI. I am interested nonetheless in solutions executing on the device if necessary.

ddibiasi, 2022-11-15:
We use the generated image to create a docker image and start it with podman. After that, you could SSH into the running system and run your checks.

But if you only want to check if a user exists in the image, you could just mount the image and look at the passwd file.

pirateradiohack, 2022-11-15:
Thanks for the tip. Will give it a try.

RiverHeart, 2022-11-15:
Depending on the complexity of your checks you don't even need to wait until the image is cooked. The "export-image" dir contains the last remaining steps before exporting an IMG file and will have the rootfs copied from the final "stage". Pi-Gen could stand to separate their substages a bit because it goes straight from "04-set-partuuid" to "05-finalise" but you can work around that by doing "040-test-image" and by setting your script to `#!/bin/bash -e` you can cancel the build for any tests that return error. Although keeping a tally of your errors and returning error at the end of the script would be more efficient.

You can look at other scripts for reference but you'll access mounted files via $ROOTFS_DIR and use "on_chroot" to run commands within the context of the chroot.