RPi-Distro/pi-gen #632: tar.xz file creation

starbasessd, 2022-09-19:
Added the line;
> DEPLOY_COMPRESSION='xz'

and the other variations listed here:
https://github.com/RPi-Distro/pi-gen#config

but it always creates a .zip file, which is not usable in the Pi-Server to create netboot in the RPi-x86 environment.
Is it broken, or just not entered correctly?

starbasessd, 2022-09-21:
Oh, sorry. Thinking (always gets me into trouble) that this is the source for RPiFoundation Files, and PiServer (part of the x86 RPiOS for AMD/Intel) the files would have been generated using this project. 
Be that as it may, The compression extension .xz is not generated by Pi-Gen correctly (even minus the .tar. part) when I select for the .xz compression method.

XECDesign, 2022-09-21:
Looks okay on this end, so some further information on how to reproduce the issue would be needed.

starbasessd, 2022-09-21:
OS:
Raspbian GNU/Linux 10 (buster) armv7l
Host: Raspberry Pi 4 Model B Rev 1.4
Kernel: 5.10.103-v7l+
Uptime: 10 days, 18 hours, 53 mins
Shell: bash 5.0.3
CPU: BCM2711 (4) @ 2.000GHz
Memory: 244MiB / 7897MiB
Using the following config file:
- - - - -
IMG_NAME='KLS'
RELEASE='buster'
LOCALE_DEFAULT='en_US.UTF-8'
KEYBOARD_KEYMAP='us'
KEYBOARD_LAYOUT='English (US)'
TIMEZONE_DEFAULT='US/Eastern'
WPA_ESSID='[redacted]'
WPA_PASSWORD='[redacted]'
WPA_COUNTRY='US'
ENABLE_SSH=1
PUBKEY_SSH_FIRST_USER="ssh-rsa [redacted]"
APT_PROXY=http://192.168.2.103:3142
FIRST_USER_NAME='pi'
FIRST_USER_PASS='[redacted]'
USE_QCOW2=0
DEPLOY_COMPRESSION='xz'
- - - - -
Need something else?

XECDesign, 2022-09-21:
I'll try an identical build tomorrow and let you know.

And are you building from head of tree (c0cddb7c834e6a0e55c11246bee40188ea50aab8)?

starbasessd, 2022-09-21:
How do I know?

XECDesign, 2022-09-21:
What does git rev-parse HEAD say?

starbasessd, 2022-09-21:
From the clone a few hours ago:
c0cddb7c834e6a0e55c11246bee40188ea50aab8
Maybe I am doing something wrong? If I do the 
>git clone https://github.com/RPi-Distro/pi-gen.git

Then have RELEASE='buster'
doesn't that give me what I want in the build?
I am confused...

starbasessd, 2022-09-22:
So how do you pull the buster build kit? The instructions are [only given for the one](git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git):
> git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git

Not all of us are gitgub wizards, and the instructions haven't changed since bullseye was added...

XECDesign, 2022-09-22:
To switch to the buster branch, you can run `git checkout buster`.

And yes, this repo is messy and incomplete.

starbasessd, 2022-09-22:
Will test shortly. Need to build a new test VM...

XECDesign, 2022-09-22:
VM? I would highly recommend building natively on a pi. qemu is likely to cause issues.

starbasessd, 2022-09-22:
I have been building on VMs for the past 2-3 years with only 1 known issue, the Certificates thing. I use both VirtualBox and Hyper-V. I use guest Debian9-i386 and Docker on other OSs. I recently was asked to look at Net Boot functionality, which is how I came across the error with .xz.  I'll deal with the no tar generated in other ways, but would like to resolve the .xz issue...
The time savings alone (less than 2 hours for a full build on a VM vs 4-6 hours on a Pi4-4GB overclocked to 2GHz) makes a huge difference to me.

starbasessd, 2022-09-22:
Get error:
> pi@rpi-i386:~/pi-gen-buster $ git checkout buster
> error: pathspec 'buster' did not match any file(s) known to git

commit 549681f6 (2022-11-08): Updates to avoid common pitfalls when changing the RELEASE variable

Fixes #635, fixes #645, fixes #632
closed 2022-11-08 by XECDesign