RPi-Distro/pi-gen #634: stage3 build fail relating to libgtk-3-0 and either galculator or gnome-calculator not installable

lbutcher5, 2022-09-27:
I am trying to build a stage3 image on a Raspberry Pi 4, of either buster or bullseye.
This issue relates to the problem I am having with bullseye.
I am starting with a fresh install from Raspberry Pi Installer, with minimum mods.
pi-gen fails just at the beginning of stage3
Is the problem the Raspberry Pi Image, the package managed by apt-get, or the pi-gen scripts?
I would appreciate any help to get the stage3 image.
What can I add, remove, or change?

There is some problem early on with libgtk-3-0, before even firing off pi-gen.
(I did apt-get update and apt-get upgrade before noticing this.)
==========
sudo apt-get install libgtk-3-0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
libgtk-3-0 : Depends: libwayland-client0 (>= 1.20.0) but 1.18.0-2~exp1.1 is to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
==========

Here is my log of everything I did to get to the Bullseye pi-gen failure.
(Targeting Bullseye works better than the Buster attempt.
I am going to open a different issue relating to Buster.)
==========
On a PC do Raspberry Pi Imager v1.7.3
Choose OS -> Raspberry PI OS (32-bit) -> 
A port of Debian Bullseye with the Raspberry Pi Desktop (recommended)
Write
(Eject)

SD Card on Raspberry Pi:
Boot into Config
  Country: United States
  Language: American English
  Timezone: Los Angeles
  Enter User Name: devel (example)
  Enter password: devel (example)
  Confirm password: devel (example)
  Skip Select WiFi Network
  Skip Update Software
  Reboot
Log in with account just created
  Raspberry Icon -> Preferences -> Raspberry Pi Configuration -> Interfaces -> VNC
From VNC Viewer, Log in with account just created
Command Window
  sudo date set=’2022-09-26 12:00’
  (Note: /etc/apt/sources.list is deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi)
  sudo apt-get update
  sudo apt-get upgrade
  sudo apt-get upgrade  (but second run reports that libgts-3-0 has been “kept back”)
  sudo apt-get install libgts-3-0
    THIS FAILS saying that it needs libwayland-client0 (>= 1.20.0) but 1.18.02-exp1.1 is to be installed
  sudo apt-get install coreutils quilt parted qemu-user-static debootstrap zerofree zip \
    dosfstools libarchive-tools libcap2-bin grep rsync xz-utils file git curl bc \
    qemu-utils kpartx gpg pigz

  mkdir build
  cd build

  git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git
  cd pi-gen
  use nano to create config with IMG_NAME=’Bullseye_Desktop’ in it
  touch stage4/SKIP stage5/SKIP
  touch stage4/SKIP_IMAGES stage5/SKIP_IMAGES
  cp stage2/EXPORT* stage3
  use nano to change “lite” to “desktop” in the stage3/EXPORT* files
Still in the build/pi-gen directory
  time sudo ./build.sh 2>&1 | tee ../build_1.log

FAILS eventually with this:

/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
                LANGUAGE = "en_US.UTF-8",
                LC_ALL = "en_US.UTF-8",
                LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
[19:56:21] End /home/sl890/build/pi-gen/stage3/00-install-packages/00-debconf
[19:56:21] Begin /home/sl890/build/pi-gen/stage3/00-install-packages/00-packages-nr
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
gcr : Depends: libgtk-3-0 (>= 3.0.0) but it is not installable
libfm-gtk4 : Depends: libgtk-3-0 (>= 3.1.4) but it is not installable
libfm-modules : Depends: libgtk-3-0 (>= 3.0.0) but it is not installable
libgcr-ui-3-1 : Depends: libgtk-3-0 (>= 3.21.5) but it is not installable
libgtksourceview-3.0-1 : Depends: libgtk-3-0 (>= 3.19.12) but it is not installable
libkeybinder-3.0-0 : Depends: libgtk-3-0 (>= 3.0.0) but it is not installable
libvte-2.91-0 : Depends: libgtk-3-0 (>= 3.24.22) but it is not installable
libwebkit2gtk-4.0-37 : Depends: libgtk-3-0 (>= 3.23.0) but it is not installable
libwnck-3-0 : Depends: libgtk-3-0 (>= 3.22.0) but it is not installable
lxde : Depends: galculator but it is not installable or
                 gnome-calculator but it is not installable
lxinput : Depends: libgtk-3-0 (>= 3.9.10) but it is not installable
lxpanel : Depends: libgtk-3-0 (>= 3.24) but it is not installable
lxterminal : Depends: libgtk-3-0 (>= 3.21.5) but it is not installable
mousepad : Depends: libgtk-3-0 (>= 3.21.5) but it is not installable
pcmanfm : Depends: libgtk-3-0 (>= 3.9.10) but it is not installable
xarchiver : Depends: libgtk-3-0 (>= 3.0.0) but it is not installable
zenity : Depends: libgtk-3-0 (>= 3.19.12) but it is not installable
E: Unable to correct problems, you have held broken packages.

lbutcher5, 2022-09-27:
Sorry about the ridiculous loud giant letters.
They look the same size in my outlook window, but cut-and-paste shows they are different.
Hmmm. Actually, the loud text is right before a line of "=" characters I used to separate sections.
Some software thinks that "=" lines means SHOUT.
Sorry.

XECDesign, 2022-09-27:
That's not a pi-gen issue, but a wrongly built package in the apt repo. Should be resolved a bit later today.

XECDesign, 2022-09-27:
Should be resolved now.

closed 2022-09-27 by XECDesign