RPi-Distro/pi-gen #635: buster stage3 build fail relating to systemd-timesyncd

lbutcher5, 2022-09-27:
This issue relates to the problem I am having with buster
I am starting with a fresh install from Raspberry Pi Installer, with minimum mods.
pi-gen fails several ways. I just keep editing things out, to get this log info
I would appreciate any help to get the stage3 image.

Sorry that this is ridiculously long.

The first failure:
[00:53:57] Begin /home/pi/build/pi-gen/stage1/03-install-packages
[00:53:57] Begin /home/pi/build/pi-gen/stage1/03-install-packages/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package systemd-timesyncd

Subsequent failure:
[01:05:58] Begin /home/pi/build/pi-gen/stage2/01-sys-tweaks/00-packages-nr
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3-picamera2

Subsequent failure:
[01:10:40] Begin /home/pi/build/pi-gen/stage2/01-sys-tweaks/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-is-python3

Subsequent failure:
[01:19:48] Begin /home/pi/build/pi-gen/stage2/02-net-tweaks/01-run.sh
'files/wpa_supplicant.conf' -> '/home/pi/build/pi-gen/work/Buster_Desktop/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/usr/bin/raspi-config: 2569: /usr/bin/raspi-config: do_netconf: not found

Subsequent failure:
[01:39:48] Begin /home/pi/build/pi-gen/export-image/01-user-rename
[01:39:48] Begin /home/pi/build/pi-gen/export-image/01-user-rename/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package userconf-pi

Here is my log of everything I did to get to the buster  pi-gen failure.

On a PC do Raspberry Pi Imager v1.7.3
Choose OS -> Raspberry PI OS (other) -> Raspberry PI OS (Legacy)
A port of Debian Buster with security updates and desktop environment
Write
(Eject)

SD Card on Raspberry Pi:
Boot into Config
  Country: United States
  Language: American English
  Timezone: Los Angeles
  Enter User Name: devel (example)
  Enter password: devel (example)
  Confirm password: devel (example)
  Skip Select WiFi Network
  Skip Update Software
  Reboot
Log in with account just created
  Raspberry Icon -> Preferences -> Raspberry Pi Configuration -> Interfaces -> VNC
From VNC Viewer, Log in with account just created
Command Window
  sudo date set=’2022-09-26 12:00’
  (Note: /etc/apt/sources.list is deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi)
  sudo apt-get update

  sudo apt-get install coreutils quilt parted qemu-user-static debootstrap zerofree zip \
    dosfstools libarchive-tools libcap2-bin grep rsync xz-utils file git curl bc \
    qemu-utils kpartx gpg pigz

  mkdir build
  cd build

  git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git
  cd pi-gen
  use nano to create config containing
  IMG_NAME=’Bullseye_Desktop’
  RELEASE=’buster’
  touch stage4/SKIP stage5/SKIP
  touch stage4/SKIP_IMAGES stage5/SKIP_IMAGES
  cp stage2/EXPORT* stage3
  use nano to change “lite” to “desktop” in the stage3/EXPORT* files
Still in the build/pi-gen directory
  time sudo ./build.sh 2>&1 | tee ../build_1.log

FAILS eventually with this:

[00:53:57] Begin /home/pi/build/pi-gen/stage1/03-install-packages
[00:53:57] Begin /home/pi/build/pi-gen/stage1/03-install-packages/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package systemd-timesyncd

edited stage1/03-install-packages/00-packages to remove systemd-timesyncd

Runs some more, then fails with:

[01:05:58] Begin /home/pi/build/pi-gen/stage2/01-sys-tweaks/00-packages-nr
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3-picamera2

edited stage2/01-sys-tweaks/00-packages-nr to remove python3-picamera2

Runs some more, then fails with:

[01:10:40] Begin /home/pi/build/pi-gen/stage2/01-sys-tweaks/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-is-python3

edited stage2/01-sys-tweaks/00-packages to remove python-is-python3

Runs some more, then fails with:

[01:19:48] Begin /home/pi/build/pi-gen/stage2/02-net-tweaks/01-run.sh
'files/wpa_supplicant.conf' -> '/home/pi/build/pi-gen/work/Buster_Desktop/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/usr/bin/raspi-config: 2569: /usr/bin/raspi-config: do_netconf: not found

edited stage2/02-net-tweaks/01-run.sh to comment out do_netconf
I DO NOT KNOW IF THIS WILL WORK WHEN I REMOVE THIS

Runs some more, then fails with:

[01:39:48] Begin /home/pi/build/pi-gen/export-image/01-user-rename
[01:39:48] Begin /home/pi/build/pi-gen/export-image/01-user-rename/00-packages
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package userconf-pi

At this point, I gave up.

XECDesign, 2022-09-27:
If you're building a buster image, you'll need to do that off the `buster` branch. Buster and Bullseye releases have different packages and setups.

lbutcher5, 2022-09-27:
I thought that is done in the config file.
RELEASE=’buster’

commit 549681f6 (2022-11-08): Updates to avoid common pitfalls when changing the RELEASE variable

Fixes #635, fixes #645, fixes #632