RPi-Distro/pi-gen #637: Is it somehow already possible to switch from kernel 5.15.61 to 5.19?

capiman, 2022-10-05:
In the past we already found a security leak which seems it was fixed in 5.19 (don't know if backported to a 5.15.x).
And today we found an error with Bluetooth which seems to be fixed with 5.17.
So anything newer than 5.17 would be helpful...

XECDesign, 2022-10-07:
That's not a feature of pi-gen, no. But, if you know what you're doing you could provide a custom kernel.

capiman, 2022-10-07:
Thanks for your answer! In general I know how to compile a kernel.
Do you know where I can get all the switches which were used to compile a kernel, so I can use similar ones to the provided one?

XECDesign, 2022-10-07:
This should give you something very close to the official kernel:

https://www.raspberrypi.com/documentation/computers/linux_kernel.html

closed 2022-11-08 by XECDesign