RPi-Distro/pi-gen #639: mkdosfs erroring due to losetup -o incorrectly creating partitions

rfc6919, 2022-10-20:
running pi-gen on a `5.18.0-0.deb11.4-arm64` system (Debian bullseye+backports, on non-RPi arm64 hardware), the losetup invocation for `BOOT_DEV` in `export-image/prerun.sh` is for some reason creating `/dev/loop0p{1,2}` even though `-P` isn't being passed to losetup. As a result, mkdosfs errors out like so:

`mkdosfs: Partitions or virtual mappings on device '/dev/loop0', not making filesystem (use -I to override)`

I'm not sure why the partitioned loop device is being created (this is presumably not a pi-gen issue), but is there any particular reason the image creation uses 2 loop devices with `-o` and `--sizelimit` specified rather than just a single partitioned loop device?

invocation, running from a checkout of `2bc1f64` on the `origin/arm64` branch. no `config` is present:

```
sudo IMG_NAME=rpi-test-lite STAGE_LIST="stage0 stage1 stage2" DEPLOY_COMPRESSION=none ./build.sh
```

failure:

```
... build output ...
[02:25:15] Begin /scratch/pi-gen/pi-gen/export-image
[02:25:15] Begin /scratch/pi-gen/pi-gen/export-image/prerun.sh
Mounting BOOT_DEV...
Mounting ROOT_DEV...
/boot: offset 4194304, length 268435456
/:     offset 272629760, length 1736441856
mkdosfs: Partitions or virtual mappings on device '/dev/loop0', not making filesystem (use -I to override)
```
```
debian@builder:/scratch/pi-gen/pi-gen$ sudo losetup -a
/dev/loop1: [0077]:194119 (/scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img), offset 272629760, sizelimit 1736441856
/dev/loop0: [0077]:194119 (/scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img), offset 4194304, sizelimit 268435456
debian@builder:/scratch/pi-gen/pi-gen$ grep loop /proc/partitions 
   7        0     262144 loop0
 259        0     262144 loop0p1
 259        1    1695744 loop0p2
   7        1    1695744 loop1
debian@builder:/scratch/pi-gen/pi-gen$ ls -lF /scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img
-rw-r--r-- 1 root root 2009071616 Oct 20 02:25 /scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img
debian@builder:/scratch/pi-gen/pi-gen$ sudo blkid /dev/loop*
debian@builder:/scratch/pi-gen/pi-gen$ sudo blkid /scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img 
/scratch/pi-gen/pi-gen/work/rpi-test-lite/export-image/2022-10-20-rpi-test-lite-lite.img: PTUUID="e2d0008f" PTTYPE="dos"
```

so it looks like `loop0` is correctly just the first partition in the image, not the entire image file, but `loop0p1` and `loop0p2` are also being created for the first and second partitions respectively.

rfc6919, 2022-10-20:
yep, the p1 and p2 partition nodes don't exist prior. simple reduction:
```
debian@builder:/tmp$ sudo losetup -D
debian@builder:/tmp$ grep loop /proc/partitions
debian@builder:/tmp$ rm -f zzz.img
debian@builder:/tmp$ dd if=/dev/zero of=zzz.img bs=1 count=0 seek=$((1024*1024*1024))
0+0 records in
0+0 records out
0 bytes copied, 0.000112496 s, 0.0 kB/s
debian@builder:/tmp$ printf '%s %s %s %s\n' - 256M c - - - 83 - | /sbin/sfdisk zzz.img 
Checking that no-one is using this disk right now ... OK

Disk zzz.img: 1 GiB, 1073741824 bytes, 2097152 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0x663a4517.
zzz.img1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 256 MiB.
zzz.img2: Created a new partition 2 of type 'Linux' and of size 767 MiB.
zzz.img3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x663a4517

Device     Boot  Start     End Sectors  Size Id Type
zzz.img1          2048  526335  524288  256M  c W95 FAT32 (LBA)
zzz.img2        526336 2097151 1570816  767M 83 Linux

The partition table has been altered.
Syncing disks.
debian@builder:/tmp$ sudo losetup -f -o $((2048*512)) --sizelimit 256M --show zzz.img 
/dev/loop0
debian@builder:/tmp$ grep loop /proc/partitions 
   7        0     262144 loop0
 259        0     262144 loop0p1
 259        1     785408 loop0p2
debian@builder:/tmp$ 
```

so the underlying issue is clearly something with `losetup` and not `pi-gen`'s problem.

but, is there any particular reason for using 2 loop devices with offset/size into the image file rather than a single loop device with partitioning? would you take a PR updating `pi-gen` to just use a single loop device? looks like it'd just be some tweaks to `export-image/prerun.sh`, `export-noobs/prerun.sh`, and `scripts/common:unmount_image()`.

XECDesign, 2022-10-21:
> is there any particular reason for using 2 loop devices with offset/size into the image file rather than a single loop device with partitioning?

No, it's only because the -P option was not widely available back when pi-gen was written.

> Would you take a PR updating pi-gen to just use a single loop device?

Yeah, it would be appreciated.