RPi-Distro/pi-gen #640: Armboost for Bitcoin Node & serveur

HotCrypt, 2022-10-21:
Hello, I have a raspberry 4b 8go, 2TB with 2xhub usb.

I run a bitcoin node (db cache 4go) and two other servers connected to my bitcoin daemon 24/24h 7/7. 

I noticed that the frequency is mostly at 1800Mhz which is normal, armboost is ok. 

I was wondering if it’s not a problem 24 hours a day. Once, a server bugged and used a lot of cpu resources, I was 100% on the cpu and it activated the power shortage alert on the power supply.

Can I do something without affecting the fluidity of my rasp too much?

Thank you for your work. 

Sincerely,

XECDesign, 2022-10-26:
This repo is for reporting issues with pi-gen. For general support and discussion, I'd recommend [the forum](https://forums.raspberrypi.com/).

But yes, as long as you don't draw more than 1.2A through the USB ports and use the official 3A power supply, you should be okay.

closed 2022-10-26 by XECDesign

HotCrypt, 2022-10-26:
> This repo is for reporting issues with pi-gen. For general support and discussion, I'd recommend the forum.
> 
> But yes, as long as you don't draw more than 1.2A through the USB ports and use the official 3A power supply, you should be okay.

Ok thanks bro